CR6Community/Marlin #248: Nozzle temperature sometimes cannot reach target - hovers instead about 10 degrees lower until "Heating Failed" timeout

Thinkersbluff, 2021-04-19:
### Description

The nozzle temperature sometimes fails to reach the target value when heating.
When this is happening, the temperature may climb to within a few degrees of target, but then drops again, cycling around a center value approximately 10 degrees below target.  Eventually, the system throws a yellow "Heating Failed" message on the screen and "kills" the job, forcing the user to cycle power to recover.

### Steps to Reproduce

I was able to reproduce this problem fairly consistently as follows:

0. Connect Octoprint to the printer & monitor the serial interface and the Temperature plot to observe what happens
1. Use the CF6 PID function to PID the nozzle at 235C 
2.  Wait for the nozzle to cool to 145C (or set the temperature to 145C, to let it stabilize there.)
3. Use the CF6 esteps function to run a calibration extrusion at 235C.
4. The printer recognizes and reports on the serial interface that the target temperature is 235
5. The printer stops short of 235 when heating, instead cooling again around 230C
6. The printer then cycles around approx 225 +/-5 degrees, never trying to achieve 235, as seen in the attached logs.
7. If left long enough (sorry, did not time it), the printer throws a Heating Failed alert and kills the process.

**Expected behavior:** 

At step 5 or step 6, the printer should overshoot 235, and gradually settle on to it, after a ring or two.

**Actual behavior:** 

The printer never gets to the target value.

**Logging:**

[image]
[wtf.txt](https://github.com/CR6Community/Marlin/files/6333196/wtf.txt)
[image]
[wtf2.txt](https://github.com/CR6Community/Marlin/files/6333208/wtf2.txt)

### Additional Information

Hardware = Stock Creality Hot End, BTT SKR CR6 v1.0 Motherboard, Stock Creality TFT, Octoprint on RPi3B+

Running this CF6 Pre2 as offered in Assets section, with no mods:
[CF6.1-Pre2-btt-skr-cr6-with-stock-creality-tft-2021-04-18-22-12.zip](https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-6.1-pre2/CF6.1-Pre2-btt-skr-cr6-with-stock-creality-tft-2021-04-18-22-12.zip)

Sebazzz, 2021-04-19:
Can you please report this upstream? I try to stay clear of thermal control code.

Thinkersbluff, 2021-04-19:
I am surprised that noone upstream seems to have noticed this issue.  It went so well last time, when you were able to show them the error in their code.  I had hoped you could see this one, too.

I will give it a go & see what reaction I get.

Sebazzz, 2021-04-19:
Maybe this has something to do with that temperature is now measured in whole integers instead of floats.

Met vriendelijke groet,
Sebastiaan Dammann

Talha909, 2021-04-19:
I have the same issue. I am using 12 April bug fix. Yesterday I try to do PID but its fails. And it does reach to desire temp. It always below 10 degree.

Thinkersbluff, 2021-04-19:
> I have the same issue. I am using 12 April bug fix. Yesterday I try to do PID but its fails. And it does reach to desire temp. It always below 10 degree.

@Talha909 Are you using the Community Firmware on a CR6 or are you saying you have a problem running 12 April 2021 Marlin bugfix2.0 on something else?

Thinkersbluff, 2021-04-19:
> Maybe this has something to do with that temperature is now measured in whole integers instead of floats. 

I am suspicious of the 10 degrees part.
Can you tell in the code whether this code in Temperature.cpp is causing PID to turn off the heater when the nozzle gets within 10 degrees of target?

if (temp_hotend[ee].target == 0
          || pid_error < -(PID_FUNCTIONAL_RANGE)
          || TERN0(HEATER_IDLE_HANDLER, heater_idle[ee].timed_out)
        ) {
          pid_output = 0;
          pid_reset[ee] = true;

If the integer roundoff error were to "trick" PID into turning off the heater too soon, I could understand it cooling when it should be heating.  It is beyond my "skills" with software & logic to decide whether that Functional Range parameter (defined as 10 in Configuration.h) could cause the PID routine to settle 10 degrees short of target if round-offs "went the wrong way" during the heating cycle.  I know that last test I ran, the PID suddenly went up and achieved target when it had been looking like it would settle 10 deg lower...

  #define PID_FUNCTIONAL_RANGE 10 
// If the temperature difference between the target temperature and the actual temperature
// is more than PID_FUNCTIONAL_RANGE then the PID will be shut off and the heater will be set to min/max.

I am away from my system today, but am tempted to try redefining that parameter to something like 5 and seeing what happens.

Sebazzz, 2021-04-19:
Workaround for now appears to be:

- Reset to factory settings and use default PID values *or*
- Increase the temperature in the leveling settings.

Thinkersbluff, 2021-04-22:
@Sebazzz - I ran some tests today with PID_FUNCTIONAL_RANGE set to 5 instead of 10.

Does it tell you anything important that I was now able to complete a full eSteps calibration, although the temperature never stabilized at 235? (i.e.: Instead of throwing a "Heating Failed" error message, the printer proceeded with the extrusion, then beeped and asked me to measure the remaining filament. )

[image]
[image]
[image]
[Test4_LOG.txt](https://github.com/CR6Community/Marlin/files/6355612/Test4_LOG.txt)

[BTW - esteps calibration ignores the filament runout sensor, because I had no filament in the machine & the runout sensor light was out, but the printer said nothing.]

Sebazzz, 2021-04-22:
My guess is this bypasses the PID because it now looks at the temperature hysterisis setting instead. However, this is not desired because this will cause terrible Z-banding with some filaments.

Met vriendelijke groet,
Sebastiaan Dammann

[pfr=5_Tests1-3]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F36551518%2F115658236-6421cc80-a306-11eb-83a2-bf3ef179b222.png&data=04%7C01%7C%7C6b9f53763bf94048225208d9054bf999%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637546646128385891%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=B%2F5hBLFJPj1zvOz%2FtOgZi6CM9rg9cLoatHHESRB9A2s%3D&reserved=0>
[pfr=5_Tests1-3]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F36551518%2F115658260-76036f80-a306-11eb-9f91-6eac2e8fbfff.png&data=04%7C01%7C%7C6b9f53763bf94048225208d9054bf999%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637546646128395888%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=waOTYLNofklqBdD3jzsY%2FwWK2i6MH2G16wRNkgaheLw%3D&reserved=0>
[pfr=5_Test4]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F36551518%2F115658266-7bf95080-a306-11eb-9e4a-447751d755f0.png&data=04%7C01%7C%7C6b9f53763bf94048225208d9054bf999%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637546646128405879%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=9JRUsJef3XdDBAFKsyUMM6kAo2W8SVpVAXt6ABtBLwU%3D&reserved=0>

Thinkersbluff, 2021-04-22:
Here is a second series using PID_FUNCTIONAL_RANGE = 2.
In both tests, the printer threw a Heating Failed warning.

[image]
[PID_FUNCTIONAL_RANGE=2_Test1-_LOG.txt](https://github.com/CR6Community/Marlin/files/6355742/PID_FUNCTIONAL_RANGE.2_Test1-_LOG.txt)
[PID_FUNCTIONAL_RANGE=2_Test2-_LOG.txt](https://github.com/CR6Community/Marlin/files/6355743/PID_FUNCTIONAL_RANGE.2_Test2-_LOG.txt)

Thinkersbluff, 2021-04-22:
In stark contrast to the above demonstrations, PID works correctly in CF6 Prel 6 (pre Marlin bugfix2.0 commit "Add typedef celsius_t #21374"):

[image]
[Before Bug#248.txt](https://github.com/CR6Community/Marlin/files/6359088/Before.Bug.248.txt)

What a difference a few significant figures after the decimal place can make...

Sebazzz, 2021-04-22:
You might want to add that to the upstream bug.

Met vriendelijke groet,
Sebastiaan Dammann

Thinkersbluff, 2021-04-22:
> You might want to add that to the upstream bug. Met vriendelijke groet, Sebastiaan Dammann

Yep. Working on it.  Using "Compare" to try to identify all of the bugfix2.0 commits that differentiate Pre6 from 6 Final.
Do you already have that list compiled or available somewhere?

Sebazzz, 2021-04-22:
You say the issue isn't there in pre6 right? 

[Comparison between pre6 and final](https://github.com/CR6Community/Marlin/compare/v2.0.7.3-cr6-community-release-6-pre6...v2.0.7.3-cr6-community-release-6)

[Comparison between pre6 and 6.1 pre2](https://github.com/CR6Community/Marlin/compare/v2.0.7.3-cr6-community-release-6-pre6...v2.0.7.3-cr6-community-release-6.1-pre2)

Thinkersbluff, 2021-04-22:
> You say the issue isn't there in pre6 right?
> 
It is not there in rel6 pre6
It is there at rel6.1 pre2
Users are reporting issues at rel6 final.  I will go test rel6 final with esteps at 235 and return to update this comment

Sebazzz, 2021-04-22:
@Thinkersbluff Yes, the Github compare URL just wants two git commit IDs or references to commit ids like branch names, tags etc.

Sebazzz, 2021-04-22:
On my first Kickstarter CR-6 with "CF7-Nightly" with distribution date "2021-03-27" I have this graph:

[image]

Unfortunately, I can't check when I compile the firmware. For now, assume this was a pre-temperature fix build.

Sebazzz, 2021-04-22:
> Is it worth it to you to create a test fork of CF6 that only incorporates that one PR#21374 into Rel6Pre6, to try to prove this theory?

I can do the git magic and push a temporary branch.

Thinkersbluff, 2021-04-22:
> > Is it worth it to you to create a test fork of CF6 that only incorporates that one PR#21374 into Rel6Pre6, to try to prove this theory?
> 
> I can do the git magic and push a temporary branch.

I think that would help us corroborate or disprove this theory.  I think it would help to be able to say that was the only difference between bug and no bug.

Sebazzz, 2021-04-22:
@Thinkersbluff Pushed a temp/celcius_t branch:

[image]

If you happen to use a command line git client:

```
git fetch origin
git switch temp/celcius_t
```

ztakis, 2021-04-22:
13 confilcted files. Should I do it?

[image]

Sebazzz, 2021-04-22:
You are trying to merge the branch in your local branch. Instead, you need to _switch_ to the branch.

ztakis, 2021-04-22:
Ahh yes. Thank you

Thinkersbluff, 2021-04-22:
There's this universal "law" that my British parents called, ["Sod's Law".](https://g.co/kgs/P4HKod)
I think I just proved that my most recent test method needs revision. (Or this may suggest that the bug is further upstream than we thought...)

I just got this result from testing Rel 6 Pre 6:
[image]
[bug248-rel6Pre6.txt](https://github.com/CR6Community/Marlin/files/6360324/bug248-rel6Pre6.txt)

Maybe this demonstrates the Facebook Group's situation, in which a user claims if he does a PID with CF6, he sees this bug, but if he resets to factory settings, his problem goes away...

I think the PID function generates p, i & d based on what it had to do to get the nozzle to achieve & stay at target temp.
If the nozzle is already close the the target temp (in my test case it was already at target temp) then the resulting parameters may not be aggressive enough to get PID to work when starting from a much lower temp...

I do not know, yet, whether this is a red-herring generated purely by a bad test procedure or whether this is another useful clue.

Thank you for the custom build.
I hate using command lines...  I have Windows  GitHub Desktop.  I see the temp/celcius_t branch.  It says I will lose some changes here if I go, so I will update this Comment when I have accessed the build.  Thank you.

Sebazzz, 2021-04-22:
@Thinkersbluff You are using the BTT SKR board right? Make sure the firmware was flashed (firmware.bin renamed to firmware.cur). On a stock Creality board, I would change the version in Version.h and check afterwards.

Thinkersbluff, 2021-04-22:
Thanks, yes.  I wish it was only that...
NOTE: The system is doing an excellent job of keeping me straight on all this flashing. (Sooo glad I put that extension cable into the display, way back when..)  I get excellent feedback between the flashing led and the messaging, to ensure that I have the right builds installed in every configuration.

Thinkersbluff, 2021-04-22:
Testing is now showing me that the problem I am able to consistently reproduce is present in **both** Rel 6 Pre 6 **and** Rel 6 Pre 6 + PR #21374.

The perpetual temperature cycling shown in this graph is consistent with the behaviour that some users are describing as, "hovering about 10 degrees below the target temperature".

[image]
[custombuildTest5.txt](https://github.com/CR6Community/Marlin/files/6361062/custombuildTest5.txt)

It appears as though the PID routine in the latest bugfix2.0 of Marlin is not generating "effective" PID parameters.
Although my earlier testing seemed to show that this bug was not present in Pre 6 Rel 6, I must now revise that assessment in light of these results:

[image]
[Rel6Pre6-Test5.txt](https://github.com/CR6Community/Marlin/files/6361306/Rel6Pre6-Test5.txt)

Thinkersbluff, 2021-04-22:
This upstream issue seems to be related: https://github.com/MarlinFirmware/Marlin/issues/20463

Thinkersbluff, 2021-04-22:
@Sebazzz - I notice that the default PID parameters for the Creality and BTT board machines are different in the CF6 configs, although their stock hotends would be the same.  Is this intentional?

[image]

Thinkersbluff, 2021-04-23:
Based on this data, I believe the problem here is that generating the right PID values is not as simple as entering a target temp and pushing the button.  I do not yet understand how Marlin decides what parameters to recommend, but I do know that the numbers I am getting with PID E0 S245 C8 U1 are definitely not the right ones.

[image]
[CustomBuild_PID DEBUG_245C.txt](https://github.com/CR6Community/Marlin/files/6362438/CustomBuild_PID.DEBUG_245C.txt)

If anyone else already understands the Classic PID algorithms and/or has a standalone commercial PID system with which they can generate the "right" values here, that would be great.  Otherwise, I am packing for a long long long deep dive down a rabbit hole...

Thinkersbluff, 2021-04-23:
After reviewing this work and recommendations uploaded by descipher on the Marlin bug #20463, I ran a couple of test with two modifications to the Configuration.h settings:
>The smoothing factor is highest as the value approaches 0.5 if you prefer to see a high level of smoothing try around 0.55 to 0.45. You can expect a slightly slower ramp up to the setpoint but a high level of stability.
My best results are with
PID_K1 0.52
PID_FUNCTIONAL_RANGE 16
EXTRUDE_MINTEMP 160
30W 12V heater, ATMEGA2560 RAMPS 1.4, 100k Thermistor
Seeing +1.5 -1.5 on the startup ripple, with stable follow on +- 0.20 fluctuations via PID_DEBUG output.

I have made no attempt to optimize these two parameters (PID_K1 & PID_FUNCTIONAL_RANGE), but preliminary testing on my system does suggest that the "bug" we are chasing here may be addressable by tuning the Configuration.h file.  
For reference - the Configuration.h file for BTT SKR CR6 with stock display contains:
   PID_K1 = 0.95; PID_FUNCTIONAL_RANGE = 10; Kp: 14.3200 Ki:0.8100 Kd:63.1200

[image]
[image]
[CustomBuild_PID SMOOTHING pt55.txt](https://github.com/CR6Community/Marlin/files/6367346/CustomBuild_PID.SMOOTHING.pt55.txt)
[image]
[CustomBuild_PID SMOOTHING pt55+RANGE16.txt](https://github.com/CR6Community/Marlin/files/6367349/CustomBuild_PID.SMOOTHING.pt55%2BRANGE16.txt)

Thinkersbluff, 2021-04-23:
In these tests, I evaluated the ability of my printer to heat and stabilize the nozzle at three different print temperatures (235C, 205C and 180C), using the same set of PID parameters.  

[image]
[CustomBuild_PID SMOOTHING pt55+RANGE16+BTTpids.txt](https://github.com/CR6Community/Marlin/files/6368558/CustomBuild_PID.SMOOTHING.pt55%2BRANGE16%2BBTTpids.txt)
[image]
[CustomBuild_PID SMOOTHING pt55+RANGE16+bestpids.txt](https://github.com/CR6Community/Marlin/files/6368559/CustomBuild_PID.SMOOTHING.pt55%2BRANGE16%2Bbestpids.txt)
[image]
[CustomBuild_PID SMOOTHING pt55+RANGE16+CRpids.txt](https://github.com/CR6Community/Marlin/files/6368561/CustomBuild_PID.SMOOTHING.pt55%2BRANGE16%2BCRpids.txt)

Sebazzz, 2021-04-24:
@Thinkersbluff Thank you for your extensive testing. Can you describe your testing procedure (or I've missed it)?

Upstream PR [https://github.com/MarlinFirmware/Marlin/pull/21678] has flowed down to this codebase, so it might be worth re-testing. Make sure you're on the extui branch (if you're still on the temp branch) and then pull in the latest changes.

Thinkersbluff, 2021-04-24:
> @Thinkersbluff Thank you for your extensive testing. Can you describe your testing procedure (or I've missed it)?
> 
> Upstream PR [MarlinFirmware#21678](https://github.com/MarlinFirmware/Marlin/pull/21678) has flowed down to this codebase, so it might be worth re-testing. Make sure you're on the extui branch (if you're still on the temp branch) and then pull in the latest changes.

The test procedures and logged data are in the text file below each graph.  It took me a bit of trial and error to settle on a consistent approach. The PID tests show that Marlin’s Autotune is not delivering “good” parameters with the current default PID_K1 and PID_FUNCTIONAL_RANGE parameters.  Changing them both makes the Autotune converge smoothly to target but leaves minor ringing. 

I will try the latest extui later. All of these tests were done with the integer temperature values, which likely smooths that ringing, but I expect we still need to address those two PID parameters.

Sebazzz, 2021-04-24:
@Thinkersbluff With the latest merges it is even more unstable 😐

Thinkersbluff, 2021-04-25:
I think we have to consider that the results I am posting here may only be valid for machines with my configuration.  They may even only be valid for my machine.   I did not have any failures while running my specific tests.  These tests take a lot of time and effort to execute and document, so I am hoping that other users will be willing and able to run similar tests on their systems and give feedback if they have problems.
Based on these tests, though, recognizing that I can not explain the results I am seeing, only present them, I am encouraged to think that one feasible work-around that would allow us to proceed with Release 6.1 testing would be to make these two changes in Configuration.h:
1. Change PID_K1 smoothing factor from 0.95 to 0.55
2. Change PID_FUNCTIONAL_RANGE from 10 to 16

Here are the plots and data that lead me to recommend that.  Once again, the test procedure and the logged results are in the .txt file below the related plot:

[image]
[6.1Nightly24Apr21_PID_K1-95_RANGE-10_DefaultKs.txt](https://github.com/CR6Community/Marlin/files/6371099/6.1Nightly24Apr21_PID_K1-95_RANGE-10_DefaultKs.txt)
[image]
[6.1Nightly24Apr21_PID_K1-55_RANGE-16_DefaultKs.txt](https://github.com/CR6Community/Marlin/files/6371100/6.1Nightly24Apr21_PID_K1-55_RANGE-16_DefaultKs.txt)

Sebazzz, 2021-04-30:
@Thinkersbluff Please check this - possibly related https://github.com/MarlinFirmware/Marlin/pull/21743

renegadecitizen, 2021-04-30:
I just got this error
On stock firmware too (I haven’t yet updated to the community firmware)
I fixed it by turning the hotend fan off
Lol
Crazy I know, but it works
Maybe you can explain this?

Thinkersbluff, 2021-05-02:
Today I made time to test PID and Autotune in Release 6.1, Prerelease 3.  I waited too long before capturing the first plot & lost the initial Autotune run with the default PID values, but it did look the same as the curve in the center of the graph.  I am not getting Heating Failed warnings any more, but the Autotune is still clearly undershooting twice before achieving the target temperature.
[image]
[image]
[6.1Pre3_PID_K195+RANGE10.txt](https://github.com/CR6Community/Marlin/files/6410606/6.1Pre3_PID_K195%2BRANGE10.txt)

maximustiberius, 2021-05-06:
I'm having the same problem.

Just updated from stock firmware to CR6Comm-Rel_Final... and getting undershooting of nozzle temperature after PID calibrating... Just about worked the second time after turning off fan for PID.
[image]

Thinkersbluff, 2021-05-06:
> I'm having the same problem.
> 
> Just updated from stock firmware to CR6Comm-Rel_Final... and getting undershooting of nozzle temperature after PID calibrating... Just about worked the second time after turning off fan for PID.

@maximustiberius - Your curves look similar to those I reported here:
https://user-images.githubusercontent.com/36551518/115781148-00d87e80-a388-11eb-9c95-a50fdc53de80.png

I have also been doing PID with the fan off.

I have a couple of questions for you that might help the Marlin developers troubleshooting this bug.
 - Which motherboard is now in your printer?
 - Are you still using the stock CR6 hot end originally delivered with the printer, or have you replaced or modified your hot end?
 - Did you try this same test using the stock firmware? (PID could be commanded by sending M303 S240 E0 U1 via the terminal.)
 - What happened if you did test the stock firmware?
 - Which version of the stock firmware did you test? (The Creality firmware is also based on Marlin, but it is a much earlier version than the Community Firmware.  It might help to reveal how far back this bug was introduced into Marlin.) 
 - If you do a factory reset of the PID values, what does the nozzle temperature curve look like, when you try printing at 240?  (There may be two separate issues, here - 1. The PID may not calculate "good" numbers. 2. Marlin PID Autotune may not converge well at target temperatures, even with "known good" values.)
 
I am curious what K values your PID routine generates.  Notice that I report my values on the graphs, in case there is a clue in that information.

maximustiberius, 2021-05-06:
> > I'm having the same problem.
> > Just updated from stock firmware to CR6Comm-Rel_Final... and getting undershooting of nozzle temperature after PID calibrating... Just about worked the second time after turning off fan for PID.
> 
> @maximustiberius - Your curves look similar to those I reported here:
> https://user-images.githubusercontent.com/36551518/115781148-00d87e80-a388-11eb-9c95-a50fdc53de80.png
> 
> I have also been doing PID with the fan off.
> 
> ## I have a couple of questions for you that might help the Marlin developers troubleshooting this bug.
> * Did you also get a Heater Failed warning/shut down event on that effort where the temperature never achieved target?

i got the heater failed warning a previous time, but in this graph, I cancelled the print!

> * Which motherboard is now in your printer?
4.5.2

> * Are you still using the stock CR6 hot end originally delivered with the printer, or have you replaced or modified your hot end?
Stock

> * Did you try this same test using the stock firmware? (PID could be commanded by sending M303 S240 E0 U1 via the terminal.)
yes... no problem with PID on Octoprint terminal

> * What happened if you did test the stock firmware?
worked fine

> * Which version of the stock firmware did you test? (The Creality firmware is also based on Marlin, but it is a much earlier version than the Community Firmware.  It might help to reveal how far back this bug was introduced into Marlin.)
Latest Creality firmware for CR6

> * If you do a factory reset of the PID values, what does the nozzle temperature curve look like, when you try printing at 240?  (There may be two separate issues, here - 1. The PID may not calculate [...]
I will try this

> 
> I am curious what K values your PID routine generates. Notice that I report my values on the graphs, in case there is a clue in that information.
I don’t know... where do you see those on the screen command? I’ll look.

thanks,

paul

Thinkersbluff, 2021-05-06:
@maximustiberius 
- Thank you for the quick replies.  It is useful to know that the 4.5.2 motherboard also has this issue.
- The K values are not displayed on the CF6 screen. They are reported via the serial interface during and at the end of the PID process itself.  Also, if you send an M301 command by itself, Marlin will respond with the current K values. 
- You can manually force a K  value change with M301, if you want to experiment with your own values. (https://marlinfw.org/docs/gcode/M301.html)

If you do try testing the Creality stock firmware, it would be interesting to know what their default PID values are.

Sebazzz, 2021-05-07:
Since the issues are core to Marlin - let's continue the discussion [in the upstream issue #21661](https://github.com/MarlinFirmware/Marlin/issues/21661). 

The issue does not seem to be necessarily related to the default configurations of the CR-6 Community Firmware. Rather, it seems to be core to something in Marlin. To aid the developers, let's continue the discussion upstream. I'Il leave this issue open however until it is resolved.

Thinkersbluff, 2021-05-07:
@Sebazzz - I recommend increasing the PID_FUNCTIONAL_RANGE parameter to 50 in Configuration.h
per [my latest comment on the upstream Marlin issue](https://github.com/MarlinFirmware/Marlin/issues/21661#issuecomment-834713632)  Worst case may be a little bit of overshoot on the initial heatup.

Sebazzz, 2021-05-07:
Alright. Note that will affect bed heating as well.

Thinkersbluff, 2021-05-07:
> Alright. Note that will affect bed heating as well.

It will, if we change the bed heater settings from Bang-Bang to PID.  Also Chamber PID and Laser Cooler, I believe.
There are different powered heaters and different thermal inertia in each case, but only the PID Kp,Ki, & Kd parameters were meant to change according to the system being controlled.  The need for any PID_FUNCTIONAL_RANGE parameter has been challenged in the past, by folks described as "knowing about PID" in their professions, as being totally unnecessary, and yet there it is.   Maybe someone expected the save processing cycles?
Values as high as 100 were recommended and the feedback was that it worked...   All it does is turn on the PID controls when the temperature is within RANGE of the target.  So yes, the BED would be much less likely than the nozzle to be outside of that PID_FUNCTIONAL_RANGE for long...

Thinkersbluff, 2022-03-12:
Hoping this upstream change is going to make things better.  https://github.com/MarlinFirmware/Marlin/pull/23871