CR6Community/Marlin #249: Issue With M600/Filament runout

Ranney1, 2021-04-20:
### Description

When using the latest firmware I run into a problem with the filament runout sensor. When triggered:

1. Print is neatly interrupted
2. Nozzle moves to park position
3. Filament is discharged neatly.
4. Screen beeps (according to M600 config) and goes to print pause screen (screen: 39_E_printresume)
5. Screens 34 and 35 are not triggered.
6. The moment the 'resume' button is pressed (screen 39_E_printresume) the filament is loaded.
7. Printer returns to start position (and starts extruding during this move).
8. At the 'resume position', extruding takes place for about 1.5 sec, after which the print is resumed.

Point 7 and certainly point 8 make the print fail.

The above is with the latest github version of Screen and Marlin.

1. Is reproducible with every print
2. Reproducible with SD print AND Octoprint
3. Reproducible with filament runout (which I mainly care about) and M600 command.
4. Have tried everything with the Config_Adv M600 config, but can't get any better.

### Steps to Reproduce

*If this is a Bug Report, please describe the steps needed to reproduce the issue*

See above.

**Expected behavior:** 

1. Filament runout triggers screens 34 and 35
2. M600 and filament runout can be used without problem from point 7 and point 8 (* see above)

**Actual behavior:** 

See above.

**Logging:**

[Enter `M111 S247` in a serial monitor (Pronterface or Octoprint). Reproduce the issue, then copy the relevant logs here. 
For Octoprint you can either user serial.log or copy/paste the output from the terminal. More information: https://marlinfw.org/docs/gcode/M111.html ]

### Additional Information

1. CR-6 SE
2. BTT Mainboard
3. Stock screen
4. Neopixels installed
5. Latest (current) github FW)
6. Problem is also with my own config. See files enclosed

My own config and config adv file in zip file (but latest community FW have same behavour)
Video of problem in zip file
Gcode example file (please wait for 1 or 2 layers and trigger filament runout. Every file gave me the same issue. This is an example file.

Maybe that i do something stupid, but after trying for days looking for help now.. Thanks!

[Github.zip](https://github.com/CR6Community/Marlin/files/6341797/Github.zip)

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]

Ranney1, 2021-04-20:
And here an serial log:

It is interesting to see what the extruder is doing in that fase. See new video in zip file.

When start moving, it is unloading fast and then loading (too far) again. After that, on the start position it is loading again.
\
[Update.zip](https://github.com/CR6Community/Marlin/files/6342039/Update.zip)

Ranney1, 2021-04-20:
And another update with more logging.

I Think that this is very interesting in the serial ;og:

[M600 ADV.txt](https://github.com/CR6Community/Marlin/files/6342145/M600.ADV.txt)

Sebazzz, 2021-04-20:
Confirmed. Happens on any gcode.

Sebazzz, 2021-04-20:
Looks like upstream commit a5d6f6ac9880090d2412bbea0027885b95c98ea2 has caused this issue. I will open an upstream bug.

Sebazzz, 2021-04-20:
Opened upstream issue: https://github.com/MarlinFirmware/Marlin/issues/21669

Sebazzz, 2021-04-20:
Opened upstream pull request with bugfix: https://github.com/MarlinFirmware/Marlin/pull/21670

Sebazzz, 2021-04-24:
Upstream commit 8b5e7429d7b874e9a9d1a6afbba74c11b36813ba has been merged into upstream, and now in downstream.

closed 2021-04-24 by Sebazzz