CR6Community/Marlin #251: Constant Thermal Runaway Errors

gerd35, 2021-04-23:
### Description

After upgrading to the newest Release (6.0) and using the PID-Tuning function for the first time, my prints constantly fail because of a thermal runaway error. At first I thought that i just forgot to add the cooling fan, resulting in a miscalibration, but turning the fan on during tuning did not resolve the issue.

Then i read issue #248 and tried resetting to default PID, which seemed to resolve it as i completed a 12 hour print afterwards. The very next print failed with a thermal runaway in the middle of it again though. These errors come very sudden, as seen in the attached image. I am aware that this may be hardware related as well, however I couldn't find much info on what might cause this and wanted to make sure that I don't needlessly buy a new hotend.
[image]

### Steps to Reproduce

1. Preheat PLA
2. Start Print via Octoprint

**Expected behavior:** 

a finished print

**Actual behavior:** 

a semi-finished print and a thermal runaway error

**Logging:**

| Send: N73658 G1 X100.142 Y128.145 E2097.50955*80
| Recv: Error:Thermal Runaway, system stopped! Heater_ID: 0
| Recv: Error:Printer halted. kill() called!
2021-04-23 19:33:12,629 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Error: Thermal Runaway, system stopped! Heater_ID: 0 - Printer halted. kill() called!"
2021-04-23 19:33:12,643 - octoprint.util.comm - INFO - Force-sending M112 to the printer
2021-04-23 19:33:12,656 - octoprint.util.comm - INFO - Changing monitoring state from "Error: Thermal Runaway, system stopped! Heater_ID: 0 - Printer halted. kill() called!" to "Offline (Error: Thermal Runaway, system stopped! Heater_ID: 0 - Printer halted. kill() called!)"
2021-04-23 19:33:12,660 - octoprint.plugins.action_command_notification - INFO - Notifications cleared

### Additional Information

Mainboard: 4.5.2
Touchscreen: Stock
Firmware: 6.0

Thinkersbluff, 2021-04-23:
Hi.  The temperature plot you have posted shows some strange events. Any idea what was happening at the time?  The issues I am chasing with #248 seem to be a result of the PID Autotune function itself not generating good values for Kp, Ki and/or Kd, so it makes sense that doing your own PID might aggravate whatever issue you are chasing.  I think the solution to #248 may turn out to be making these two changes to the Configuration.h file: 
1. Change PID_K1 from 0.95 to 0.55 (final recommended value may differ slightly)
2. Change PID_FUNCTIONAL_RANGE from 10 to 16 (again, final value may differ)
If you try these values in your system, you may find it generates better PID parameters, you may not.

If you send M301 to your printer after doing your own PID, you will be able to evaluate what impact (if any) changing those parameters has had. 

I am using a BTT SKR CR6 Board, so I cannot test those two changes on my own system.  
I would be curious to know whether they help you or not.  

I don't think your problem is related to mine, though, except that your PID Autotune function may also not generate "good" PID parameters.

It looks like you have an intermittent heater.  That final squiggly line at the most recent end of the timeline is probably the event that triggered the shutdown. (More than 4 degrees below target for at least 60 seconds)

Have you checked for "loose things" ?
Any other printing artifacts?

temperature.cpp will throw a Thermal Runaway warning during PID if the temperature drops more than 15 degrees below the target            
> else if (current_temp < target - (MAX_OVERSHOOT_PID_AUTOTUNE)) // Heated, then temperature fell too far?
              _temp_error(heater_id, str_t_thermal_runaway, GET_TEXT(MSG_THERMAL_RUNAWAY));

When printing, temperature.cpp monitors the hotend for runaway conditions, with this code:
      #if ENABLED(THERMAL_PROTECTION_HOTENDS)
        // Check for thermal runaway
        tr_state_machine[e].run(temp_hotend[e].celsius, temp_hotend[e].target, (heater_id_t)e, THERMAL_PROTECTION_PERIOD, THERMAL_PROTECTION_HYSTERESIS);
      #endif

Configuration_adv.h provides this documentation + parameters for Thermal Runaway:
/**
 * Thermal Protection provides additional protection to your printer from damage
 * and fire. Marlin always includes safe min and max temperature ranges which
 * protect against a broken or disconnected thermistor wire.
 *
 * The issue: If a thermistor falls out, it will report the much lower
 * temperature of the air in the room, and the the firmware will keep
 * the heater on.
 *
 * The solution: Once the temperature reaches the target, start observing.
 * If the temperature stays too far below the target (hysteresis) for too
 * long (period), the firmware will halt the machine as a safety precaution.
 *
 * If you get false positives for "Thermal Runaway", increase
 * THERMAL_PROTECTION_HYSTERESIS and/or THERMAL_PROTECTION_PERIOD
 */
#if ENABLED(THERMAL_PROTECTION_HOTENDS)
  #define THERMAL_PROTECTION_PERIOD 60        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 4     // Degrees Celsius

Sebazzz, 2021-04-23:
@gerd35 I would first exclude the possibility of a hardware issue. I've seen multiple cases where the thermistor wires actually became loose in the connector.

gerd35, 2021-04-24:
Thanks for your quick replys. I opened up the hotend and pushed what looked to be a little coil around a wire back into the connector of what i assume to be the thermistor (it doesn't look like the spare part from 3djake at all unfortunately, the other wire of that connector had the coil inside the connector though). After that i finished a 4h print successfully. However: I noticed that both of the last successful prints were printed from the SD card, all prints using upload from Octoprint failed with above issue. I will now attempt another 2h print using the latter method.

gerd35, 2021-04-24:
Issue seems to be resolved, thanks for the help!

closed 2021-04-24 by gerd35