CR6Community/Marlin #252: FR: Show percentage completed on Touchscreen when printing from Octoprint

gerd35, 2021-04-24:
### Description

If feasible (i don't know enough about Octoprint and this firmware to judge this) it would be nice to be able to see the percentage to which the print is completed on the touchscreen as well when printing from Octoprint, right now it says 0% until completion and the actual completion percentage is only visible in octoprint.

ritchiedc, 2021-04-24:
I just tried printing a file from octoprint using a gcode file made in Cura where I added a "Display Progress On LCD" post processing script and choosing "Percentage". The progress bar is updated on the screen.

Sebazzz, 2021-04-24:
Thank you for your inquiry. Instructions are provided in the release notes. You need to emit M73 via your slicer or a plugin. Not a bug, nor something the firmware can provide (the firmware only accepts commands but doesn't know how many are yet to come).

closed 2021-04-24 by Sebazzz