CR6Community/Marlin #254: Homing Issue after changing to BTT SKR Board

gerd35, 2021-04-30:
### Description

Me again, sorry for the troubles in advance. I recently bought the BTT SKR Board, since my printer had the 4.5.2 version and I didn't want to risk anything. It was delivered today and after some chipped fingernails and cursing because of Crealitys hot glue mania, I got it installed and flashed according to the guide at damsteen.nl . However: when trying to home the printer, it now stops about a cm before the build plate, then does its usual wiggle as if the build plate was right there and concludes homing with that. Needless to say, automatic bed levelling etc. is not possible either. 

After rereading the guide and going through the post-installation checklist again, i noticed that i couldn't move the Z-Axis in the "Move" menu anymore. I rechecked the wiring for both Z-Axis steppers and plugged them out and back in. After that, i went straight to the "Move" Menu and voila, i could move every axis perfectly again. That joy didn't last though, as i tried homing afterwards and got the same odd behaviour. After homing, i wasn't able to move the Z-Axis from bespoke menu anymore. Am i missing something?

### Steps to Reproduce

1. Install new BTT SKR Board; flash respective Motherboard firmware
2. try moving Z+/Z-
3. try homing

**Expected behavior:** 

homing as usual

**Actual behavior:** 

homing in the air, hovering 1-2cm above the build plate

### Additional Information

Motherboard : BTT SKR CR6
Firmware : 6.0, for BTT Board with Stock Screen
Touchscreen : Stock

Note: I did not reflash my touchscreen firmware as i already had 6.0 on there and figured there would be no difference. If there is and that caused the issue, i am incredibly sorry for my laziness.

Sebazzz, 2021-04-30:
Please check the #troubleshooting channels in Discord for support. The issue tracker is for actual bugs, thank you.

It is possible your hotend board defect, you can check that by issueing M119. Also check the troubleshooting section here: https://damsteen.nl/blog/2021/04/19/on-strain-gauge-leveling

closed 2021-04-30 by Sebazzz

gerd35, 2021-05-22:
I just solved the issue mentioned above, the BTT board seems to have been the culprit. However I have a new (probably hardware-related) issue now where the printer won't boot and the Discord link seems to be invalid. Has it been updated somewhere?

grobux, 2021-05-22:
About Discord, you can try this link: https://discord.com/invite/RKrxYy3Q9N

gerd35, 2021-05-23:
That seems to be the same link as in the Project Readme, it says it's invalid when I click on it..

gerd35, 2021-05-24:
Nevermind, a miraculous turn of events has it that the BTT board now works while the stock board does not. I have no reasonable explanation for this but I'm glad my troubleshooting odyssee has come to an end now.