CR6Community/Marlin #255: Z coordinate should display two decimal places

dryphi, 2021-05-01:
### Did you test the latest `extui` code or prerelease?

No, I cannot complete

### Bug Description

The cartesian coordinates displayed on the bottom of the LCD are super helpful, but they're currently set to display only one digit after the decimal (1/10th of a mm accuracy). That's fine for X and Y, however it would be preferable for the Z-height to display two decimal places, considering the Z-offset is accurate to two digits.
[image]
[image]

### Bug Timeline

New AFAIK

### Expected behavior

Z height to display two decimal places

### Actual behavior

Z height displays only one decimal place and rounds up to the nearest tenth of a millimeter

### Steps to Reproduce

1. Print anything
2. Look at bottom right of LCD screen

### Version of CR6Community Firmware

CF6

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.2

### Display

Creality CR-6 stock touch screen

### Mods

All-metal extruder

### Add-ons

N/A

### Your Slicer

Cura

### Host Software

OctoPrint

Sebazzz, 2021-05-02:
Thank you for your suggestion. Previous display of the axis positioning didn't allow the extra space for the decimal. When it was eventually redesigned, no one thought of changing it to two decimals.

closed 2021-05-02 by Sebazzz

dryphi, 2021-05-02:
Perfect thanks. Maybe for positions less than 10 there can be two
significant digits whereas when it's 10 or greater it reduces to one.
IMO this will mainly be used at the start of prints to check the z-offset
and troubleshoot adhesion issues etc.

commit 2ec32276 (2021-05-02): Show two decimal places on the Z-axis - fixes #255