CR6Community/Marlin #256: [FR] Allow editing the leveling mesh while printing

Sebazzz, 2021-05-02:
### Is your feature request related to a problem? Please describe.

Some users are, after multiple attempts, not able to get leveling right. This is caused by hardware issues which these users cannot resolve (due to the hardware or required skill level).

### Are you looking for hardware support?

N/A

### Describe the feature you want

It would be useful for these users to be able to access and modify the leveling mesh while printing their first layer. This builts on top of manual bed leveling built in #209.

Sebazzz, 2021-05-02:
(It might be able to extend this also the mesh leveling test pattern - so it can be changed while it is being printed - but I wouldn't know where to place that button @ztakis)

Sebazzz, 2021-05-02:
I've implemented a button @ztakis - Would you have time to update the PSDs? 
Note: the label on the "extra tuning" screen should actually read "Edit" instead of "View".

ritchiedc, 2021-05-02:
I was thinking about something similar the other day as I was trying to fine tune my mesh with a feeler gauge. I was constantly having to go between the view mesh screen to adjust values and the movement screen to go to difference spots. A manual assist function as part of the view mesh screen may help with this. It might work something like this: when you select a mesh location the nozzle moves to that location and then you can adjust the height similar to the z offset and when it's good per the feeler gauge you save that value. I guess you'd want to set the feeler gauge height at the home position to start with.

Sebazzz, 2021-05-02:
That is something else entirely that warrants its own feature request.

Met vriendelijke groet,
Sebastiaan Dammann

ztakis, 2021-05-03:
> 
> 
> I've implemented a button @ztakis - Would you have time to update the PSDs?
> Note: the label on the "extra tuning" screen should actually read "Edit" instead of "View".

Are you OK  with this:

ztakis, 2021-05-03:
[image]

Sebazzz, 2021-05-03:
Looks good!

ztakis, 2021-05-05:
While attempting to use this feature, got stuck in the leveling screen. Back button doesn't work and the scroller says:
"Wait for leveling completion..."

Sebazzz, 2021-06-04:
This has been fixed.

closed 2021-06-04 by Sebazzz