CR6Community/Marlin #260: [FR] Abort print at homing and heating

astralaster, 2021-05-27:
### Describe the feature you want

It is currently not possible to abort a print right after the start, because it starts in homing and heating which you cannot stop. Its not very important, but I think it would be nice to be able to abort it any time.

Sebazzz, 2021-07-07:
Sorry, this isn't possible. Marlin doesn't allow aborting homing, though we can cancel the heating process similar to how the mesh bed leveling pattern does.

BloodyRain2k, 2021-08-05:
@Sebazzz that would be at least something, given that it spends most of it's time on that.