CR6Community/Marlin #261: [FR] Manual control of extruder

astralaster, 2021-05-27:
### Is your feature request related to a problem? Please describe.

There are options for every stepper motor in the printer to run manually, but I cannot use the extruder without heating the nozzle (filament load / unload feature).

### Describe the feature you want

Add motor controls for the extruder in the prepare menu.

JamesRayn, 2021-06-08:
As a workaround maybe M302 (Cold Extrude) can help you.

https://marlinfw.org/docs/gcode/M302.html

Sebazzz, 2021-11-28:
Yes, manual extrusion support is only supported with an hot hot-end. On stock Marlin you also need to drop down to serial commands.

closed 2021-11-28 by Sebazzz