CR6Community/Marlin #262: Heating and Homing: Bed Temperature increased to 70°C instead 50°C from gcode file

MarkusP, 2021-06-22:
### Did you test the latest `extui` code or prerelease?

No, I cannot complete

### Bug Description

Created an gcode file for printing TPU. Used latest Cura Slicer (4.9.1) and Generic TPU 95A with 0,4mm Noozle as profile. Noozle temperature (printing and initial) is 230°C. Heatbed temperature (printing and initial) is 50°C.

After exporting the gcode file and selecting it from the sdcard to print it, the following happens:
- heatbed will be heated up to 50°C
- noozle will then be heated up to 230°C
- right after that the "heating up and homing in progress" screen appears and
- right that moment the heatbed will be heated up again, now to 70°C

please see video link
https://youtu.be/B3dYHCs3Fyc     (see from 2min 10sec)

Image 1
[image]

Image 2
[image]

any idea why this happens? bug? misconfiguration?
Thanks
Markus

### Bug Timeline

new bug

### Expected behavior

if temperatur of the heatbed is configured to 50°C i expect it to stay at 50°C

### Actual behavior

thru the homing and heating process the heatbed goes up to 70°C

### Version of CR6Community Firmware

firmware-20210327-155242

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.2

### Display

Creality CR-6 stock touch screen

### Mods

no

### Add-ons

no

### Your Slicer

Cura

ritchiedc, 2021-06-22:
Have you looked at your start gcode in cura?  You may want to use Sabazzz's start code at the following link:

https://damsteen.nl/blog/2021/02/11/creality-cr6-community-firmware-start-print-without-drooping

MarkusP, 2021-06-22:
Ok, thx - will give it a try and come back later...

MarkusP, 2021-06-22:
> Have you looked at your start gcode in cura? You may want to use Sabazzz's start code at the following link:
> 
> https://damsteen.nl/blog/2021/02/11/creality-cr6-community-firmware-start-print-without-drooping

Ok, that was it... worked... thanx!

closed 2021-06-22 by MarkusP