CR6Community/Marlin #263: [BUG] Side fan on the hotend is not running.

Andy81, 2021-06-29:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

During the first test print I noticed that the side fan on the hotend is not running. The display shows 0%, with the Stock-FW this started automatically. Is it intentional that it has to be set manually? Should this always run at 100%?

### Version of CR6Community Firmware

Release 6.1

### Printer model

Creality CR-6 SE

### Motherboard

Creality v1.1.0.0-ERA

### Display

Creality CR-6 stock touch screen

### Add-ons

Octoprint

### Your Slicer

Cura

### Host Software

OctoPrint

ritchiedc, 2021-06-29:
You might want to check your slicer settings. The part cooling fan (side fan) is controlled by the slicer. Cura has a number of settings related to fan speed. In my default profile the fan is off for the first layer. I've never looked into why but I suspect it improves bed adhesion and/or reduces warping by slowing down the cooling of the first layer.

Andy81, 2021-06-29:
The fan has now also jumped to 33%. Should all be fine.

What I also noticed is that the progress indicator on the display is still at 0%, Octoprint says that it is already 5%. The gcode was transmitted with Cura via the Octroprint plug-in.

Sebazzz, 2021-07-07:
You must emit M73 commands from Octoprint or the slicer, check here: https://damsteen.nl/blog/2021/03/13/estimated-time-remaining-with-the-cr6-community-firmware

closed 2021-07-07 by Sebazzz