CR6Community/Marlin #265: [FR] Printing speed change during print

thaukap, 2021-07-11:
### Describe the feature you want

It's a such a nice feature on the I3 MK3S+ to change the printing speed while the machine is printing. Hence , once it's all set up, and it only needs to go upwards, you can't make it faster. And for the first layer, get it down. The MK3 is based on Marlin, too.

Maybe I missed it here and I simply don't know how to do this.

Otherwise, the Community firmware is great. I stuck with some of the beta one, 5 or 6, but it works great. Never change a winning team. Don't need more display tweaks. 

And yes, sometimes, the firmware fails to power on correctly and it's not initialised properly. Turn it off, turn it on, usualy solves the problem. Never had these issues on the original firmware.

ritchiedc, 2021-07-11:
I think what you want is the feed rate which is on the print screen next to flow rate. The terminology comes from the corresponding marlin gcode command

thaukap, 2021-08-08:
Ritchie, thanks for the feedback. That's it what I meant. And it works. Sometimes an icon, like the speedometer, says more than words.

I'm using the Prusa slicer. It injects time remaining information. But when you bump up the feed rate, i.e. from 100% to 125%, the ETA is not recalculated. Still the one of 100%. two hours before, two hours at 125%. Instead of one and a halb. The MK3S+ does this.

closed 2021-08-08 by thaukap