CR6Community/Marlin #266: [BUG] e-step calibration calculation yields the opposite result to expected

marcos-eb, 2021-07-17:
### Did you test the latest `extui` code or prerelease?

No, there is no prerelease

### Bug Description

Downloaded 6.1 release for 4.5. motherboard.
Run e-steps calibration with settings (starting with 95 e-steps):
- extrude 200mm
- mark at 220mm
After measuring there are 28mm from the mark to the extruder, therefore the printer has understruded 8mm worth of filament. To compensate this behaviour the result of the calculation should increase the e-steps, instead of that it decreases it (and at first sight I do not realise), upon performing a second and a a third calibration I realise that the offset from my expectations and the results starts to diverge, where I realise that the performed calculation should be the other way around

### Bug Timeline

The feature did not exist on my former version of the firmware

### Expected behavior

Understrusion should result on the e-steps to increase

### Actual behavior

Understrusion resulte on the e-steps to decrease

### Steps to Reproduce

Downloaded 6.1 release for 4.5. motherboard.
Run e-steps calibration with settings (starting with 95 e-steps):
- extrude 200mm
- mark at 220mm
After measuring there are 28mm from the mark to the extruder, therefore the printer has understruded 8mm worth of filament. To compensate this behaviour the result of the calculation should increase the e-steps, instead of that it decreases it (and at first sight I do not realise), upon performing a second and a a third calibration I realise that the offset from my expectations and the results starts to diverge, where I realise that the performed calculation should be the other way around

### Version of CR6Community Firmware

6.1

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.2

### Display

Creality CR-6 stock touch screen

### Mods

None

### Add-ons

None

### Your Slicer

Other (explain below)

tomearp, 2021-08-04:
Just updated my CR-6 SE to 6.1. BTT SKR CR-6 + Creality TFT.

Restored factory settings and followed the e-step calibration process and was not able to reproduce the reported issue.

Sebazzz, 2021-09-04:
Assuming resolved.

closed 2021-09-04 by Sebazzz