CR6Community/Marlin #267: [FR] Order files by last-modified when selecting what to print

BloodyRain2k, 2021-07-26:
### Describe the feature you want

The files being ordered by last-modified when selecting what to print, is pretty much the only thing I really miss from the Creality firmware.

It's really bothersome when you're keeping all old prints on the card, just because why not, and then the new ones end up at the bottom just because they start with a late letter.

trevorade, 2022-01-20:
I'd be willing to try to help out with a PR but I'm unfamiliar with the code base and I have no idea where the file select UI even is...

Any pointers on where to look to see about implementing this behavior?

BloodyRain2k, 2023-04-02:
Another thing I noticed only recently: while printing, the UI is scrolling through long filenames like a "news ticker", but during selection of said file, it does not.
Not even in the confirmation dialog.

I can assume that such scrolling isn't that easy to do on a micro controller vs a full OS app, but if it can do so during printing, it should do so too when the confirmation is asking if this is the file the user wants to print.

Sebazzz, 2023-04-02:
No, this is not it. The scrolling and rendering is handled via the touch screen firmware. The only thing the motherboard firmware does is "give this element at this position in memory this value".

Since the positions in memory are fixed, the touch screen and motherboard firmware need to implicitly agree on what must be placed where. Otherwise all rendering is handled by the DGUS OS firmware (which uses an ASIC according to the docs).

Met vriendelijke groet,
Sebastiaan Dammann