CR6Community/Marlin #268: Smart Filament Sensor Support for Stock 4.5.2 Board

S7EN, 2021-07-27:
Is it possible to easily replace the stock filament runout sensor and attach a Bigtreetech Smart Filament Sensor on the 4.5.2 Board instead and run it with the Community Firmware?

TommyC81, 2021-08-28:
Also interested in this. Does the BTT smart sensor work as a slot-in replacement for the stock filament sensor?

jjhu, 2021-11-05:
It does seem to have the same connection. What's the added value to this sensor though? I see that it detects when the filament is broken in the sensor, the print will stop. That's also the same feature the current runout sensor has.

S7EN, 2021-11-05:
It also detects if filament is moving. When the nozzle is clogged this is very helpful.

jjhu, 2021-11-05:
Ah interesting feature, the CFW should be able to accept the BTT smart sensor hardware wise (since it's the same connected). Software wise only a few modifications in the configuration files should do the trick, https://www.youtube.com/watch?v=9gLSJIxiuwo start at 3:05.

Sebazzz, 2021-11-28:
It is supported in the sense that the underlying Marlin supports it, but it cannot be enabled dynamically. You need to compile the firmware yourself. I have written an article on this earlier: https://damsteen.nl/blog/2021/01/08/how-to-compile-cr6community-marlin-with-vscode-platformio

Good luck!

closed 2021-11-28 by Sebazzz