CR6Community/Marlin #270: [FR] Display: Remaining printtime

DuckAvenger, 2021-08-21:
### Describe the feature you want

Hi

I would be nice if you could add the remaining printtime as informatization on the display.

Thanks

ritchiedc, 2021-08-21:
The community firmware already does this via M73 messages. This can be configured in both cura 4.9+ or prusaslicer.

DuckAvenger, 2021-08-22:
Thanks @ritchiedc I found the setting in Cura 

[image]

ritchiedc, 2021-08-22:
Don't know if you noticed but the drop-down that shows M117 in your screen capture, change that to M73. This will put the remaining time underneath the print time instead of in comments in yellow on the screen. Prior to cura 4.9, M117 was the only option which was just a comment to the screen. In 4.9 they implemented the newer M73 specific for reminding time.

DuckAvenger, 2021-08-24:
Thanks @ritchiedc 

[image]

Sebazzz, 2021-11-28:
This is already supported: https://damsteen.nl/blog/2021/03/13/estimated-time-remaining-with-the-cr6-community-firmware

closed 2021-11-28 by Sebazzz