CR6Community/Marlin #274: [FR] FAN_KICKSTART_TIME 100

omexlu, 2021-08-30:
### Describe the feature you want

Hello,

The following is disabled in the configuration "//#define FAN_KICKSTART_TIME 100", but this can be beneficial for various fans.

Shouldn't this perhaps be enabled or is there perhaps a way to start this in gcode?

Thanks.

jjhu, 2021-11-05:
Considering it's the fan to cool the prints, there are multiple fillaments that don't require the fan to be at 100% or on for that matter. The hotend fan is on at start. 

Don't see the need in the machine making extra noise and wasting power when there is no active print.

Sebazzz, 2021-11-28:
@jjhu The FAN_KICKSTART_TIME feature is for shortly running the fan so it starts properly (overcome initial inertia) for low PWM values.

Sebazzz, 2021-12-19:
@jjhu FAN_KICKSTART_TIME was actually already implemented in most of the sample configs

closed 2021-12-19 by Sebazzz