CR6Community/Marlin #275: [BUG] Bed leveling: requested temperature fluctuations

keukenrol, 2021-09-08:
### Did you test the latest `extui` code or prerelease?

No, there is no prerelease

### Bug Description

When activating the auto bed leveling, the ordered temperature fluctuates between 0 and 145 about once every 2 seconds. This is causing a fluctuation in ventilation fan speed and also in actual temperature. PID calibration is done but still overshoots and undershoots a lot.

### Bug Timeline

Started when upgrading from creality V2.0.1.2 to community 6.1 (motherboard and TFT)

### Expected behavior

Stable temperature.

### Actual behavior

Unstable temperature.

### Version of CR6Community Firmware

6.1

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.2

### Display

Creality CR-6 stock touch screen

### Your Slicer

Cura

ritchiedc, 2021-09-09:
This is by design to improve abl accuracy. Others will have to enlighten us as to the reason not running the heaters during probing helps. You can change this behavior by turning off improved accuracy from the touch screen ui.

Sebazzz, 2021-11-28:
Yes, this is as intended and can be turned off in the menu.

closed 2021-11-28 by Sebazzz