CR6Community/Marlin #276: [BUG] Hotend temperature drops to 0 after homing

guivo, 2021-10-02:
### Did you test the latest `extui` code or prerelease?

No, there is no prerelease

### Bug Description

After starting a print the printer warm-up the bed and then the hotend, homes, then as soon as the probe touches the bed the hotend target temperature drops to 0.

This causes the temperature to drop and require the printer to wait until the requested temperature is reached again.

### Expected behavior

The temperature should just remain high after the target is reached.

### Version of CR6Community Firmware

6.1

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.2

### Display

Creality CR-6 stock touch screen

ritchiedc, 2021-10-02:
Same as issue #275. It is works that way by design to improve accuracy. If you don't like that behavior you can turn off better accuracy in leveling settings.

guivo, 2021-10-03:
Thanks for the reply, it worked after the first time. Is there a page where I can check the implication of the many options?

ritchiedc, 2021-10-03:
The best single place for info is the release notes but it may not answer all your questions. But no, there is no manual or site equivalent that I know of. The people that maintain this firmware volunteer their time and have real jobs and families. A manual would be very time consuming to create and maintain. If you have questions the discord server is a good place to ask.

Sebazzz, 2021-11-28:
Yes, this is as intended and can be turned off in the menu.

closed 2021-11-28 by Sebazzz

SANGERA2, 2021-12-18:
Sorry to reply to a dead thread, but if you want the increased accuracy, it's a good idea to ensure you are using M190 and M109 before you start printing after a home command in case the temp has dropped too much (probably after moving the hotend to a corner with a z of 20mm in case it drops filament onto the bed). Mine drops by 10c when homing and still isn't back up to 200c when I start to print otherwise!