CR6Community/Marlin #278: [FR] Print menu sort by feature

spmedia, 2021-10-10:
### Is your feature request related to a problem? Please describe.

Kinda.

The Print menu currently only displays the oldest file first and if you add a file, you have to scroll down thru all the files to find it

### Are you looking for hardware support?

No

### Describe the feature you want

I would like to suggesting adding the ability to the Print menu to display files by two types.

1) Sort by alphabetical file name ( A -> Z or Z -> A)
2) Sort by file date added (oldest / newest)

Right now it shows the oldest files first in the Print menu and I have to scroll all the way to the bottom to find the newest added file. I have ~80 STLs on my card so every time I want to print the newest thing added, I gotta hit the down scroll button 80 times and then hit Print. 

It would be neat if users could customize how the Print menu displays the files and be able to sort the files displays on the Print menu by file name or date added. 

This would also save users a bit time and frustration :)

spmedia, 2022-08-02:
BUMP :D 

Did file sorting ever become a thing yet?

BloodyRain2k, 2022-08-02:
I don't know if there was even an update at all yet : (