CR6Community/Marlin #284: [BUG] Screen initialisation overrides orientation

pienjo, 2021-11-13:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

The screen initialisation in ``DGUSDisplay::SetTouchScreenConfiguration`` overrides the display orientation as stored in the display firmware, causing displays that are **not** in a landscape orientation to show garbage.

### Bug Timeline

Issue has been around for a while, but the CR6-SE is not affected.

### Expected behavior

The lower two bits of the configuration byte (which stores the screen orientation) is retrieved from the display and retained. A landscape display will work correctly:

[image]

(note: this is with a modified version of ``DGUSDisplay::SetTouchScreenConfiguration`` which enfoces a 0-degree rotation - this will cause problems with CR6-SE displays!)

### Actual behavior

The lower two bits of the configuration byte (which stores the screen orientation) is overwritten with two 1's, enforcing a 270 degree orientation. A landscape display shows garbage
[image]

### Steps to Reproduce

1. Find a printer with a landscape display, and install a compatible screen firmware (e.g. my work-in-progress fork at https://github.com/pienjo/CR-6-touchscreen).
2. Observe that the display starts correctly, but shows garbage near the end of the initialisation sequence. 
3. Reboot the display by tapping the Community Edition logo a few times, and skip to the main menu by tapping the progress bar.
4. Observe that the display works correctly - until the display configuration is changed.

### Version of CR6Community Firmware

https://github.com/CR6Community/Marlin/pull/277

### Printer model

Other (I will explain below)

### Motherboard

Other (I will explain below)

### Display

Creality CR-6 stock touch screen

### Mods

Stock Creality CR200B, using the stock Creality 4.2.5 (Not 4.5.2!) board.

Sebazzz, 2021-11-28:
I see, let me check.

Sebazzz, 2021-11-28:
I've added a configuration flag DGUS_LCD_UI_CREALITY_TOUCH_ORIENTATION to Configuration.h to allow configuration of this. While I agree that getting the config from the touch screen itself it the most ideal way, it would require some rebuild of the current code paths - and I don't trust DWIN to report this correctly anyway.

I've updated the configs from PR #277 already.

closed 2021-11-28 by Sebazzz

Sebazzz, 2021-12-19:
I was just made aware of a mistake I made in the implementation @pienjo - should be alright now.

commit b9da8b71 (2021-11-28): Allow configurable screen orientation in Configuration.h - fixes #284