CR6Community/Marlin #285: Flexing and probe temp drop

omgsh, 2021-11-19:
### Did you test the latest `extui` code or prerelease?

No, I cannot complete

### Bug Description

Seeing the hot end flex the bed and gantry when probing. as if the software is trying to probe too fast or hard

 Also the temp drops 50 degrees from the start to finish of leveling providing for pretty bad corner leveling.

(happens on all my 6-se's with the 6.1)

### Bug Timeline

noticed about 2 weeks ago now

### Expected behavior

I expected the bed not to move, head not to lift off the gantry or the gantry to lift from the Z rods when leveling.

### Actual behavior

the opposite of what is directly above.

### Steps to Reproduce

Run a probe pattern to level the bed

### Version of CR6Community Firmware

6.1

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.2

### Display

Creality CR-6 stock touch screen

### Mods

nope

### Add-ons

does not need any.. its a 6se

ritchiedc, 2021-11-19:
Sebazzz (username on discord) has a good article on adjusting the strain gauge trigger weight which is not controlled by the firmware. .
 https://damsteen.nl/blog/2021/04/19/on-strain-gauge-leveling

If you don't like the temperature decrease during abl turn off the "better accuracy" option from the touch screen. This is mentioned in the release notes.

Sebazzz, 2021-11-28:
Yes, that is the way to go.

closed 2021-11-28 by Sebazzz