CR6Community/Marlin #286: [BUG] Temps changing after G28 to 200/60 instead of what is in GCode due to Better Accuracy setting

talentedmrripley, 2021-11-27:
### Did you test the latest `extui` code or prerelease?

No, there is no prerelease

### Bug Description

I recently added the GCode from this blog post (https://damsteen.nl/blog/2021/02/11/creality-cr6-community-firmware-start-print-without-drooping) to try and avoid the filament droop before printing.  Watching the behavior in OctoPrint I can see that it sets the temps correctly at first to 120/50.  The printer checks the X and Y end stops and centers the X/Y axis but then pauses and heats to 200/60 before checking Z.  After that it moves to the corner and begins warming to the temps specified by the slicer and proceeds to print.

Running a G28 from the OctoPrint terminal also heats to 200/60.  As a test I changed the Better Accuracy temps and issued another G28 and it preheated to the newly changed temps.

I can't find anything that says this is the expected behavior.  I could turn off the Better Accuracy setting or change its temps to match the no droop setting but I don't think that should be necessary and I'd prefer to leave it enabled and with the specified temps for running the leveling.

### Bug Timeline

New.  Not tested on stock firmware.

### Expected behavior

Heat to the temps specified in the GCode.

### Actual behavior

G28 causes the bed to heat to the values in the Better Accuracy settings.

### Steps to Reproduce

Run G28

### Version of CR6Community Firmware

CR6Comm-Rel6.1

### Printer model

Creality CR-6 SE

### Motherboard

Creality v1.1.0.0-ERA

### Display

Creality CR-6 stock touch screen

### Your Slicer

Cura

### Host Software

OctoPrint

Sebazzz, 2021-11-28:
Do you have a minimal gcode file that reproduces this issue? (prefer without any extrusion)

talentedmrripley, 2021-11-29:
I've included one here.  I edited it down from a PETG print I was doing.  It only heats the printer and then draws the purge lines.  But I'm not sure that it's relevant.  I tested again and simply issuing a G28 also causes the extruder and bed to heat to the Better Accuracy temps.

[Better Accuracy test.gcode.txt](https://github.com/CR6Community/Marlin/files/7619606/Better.Accuracy.test.gcode.txt)

ritchiedc, 2021-11-29:
Your original description had information that did not seem typical. If your only issue is that the nozzle and bed heat to the bed leveling temps during a g28 that is the intended operation. While not explicitly stated it is implied in the article you referenced "Preheating the nozzle before homing already happens in firmware." Admittedly it's not explicitly spelled out but when you consider both cases are probing or homing in z it makes sense why they are tied together. I think you can turn off that behavior by setting the temps to 0 but can't test it at this time since I have a print running.

The "Better Accuracy" button only affects whether or not heating is turned on while actively probing.

Sebazzz can correct me if I'm wrong.

talentedmrripley, 2021-11-29:
> Your original description had information that did not seem typical. If your only issue is that the nozzle and bed heat to the bed leveling temps during a g28 that is the intended operation. While [...]

So are you saying that a G28 should heat the bed and extruder to the specific temperatures set in the Better Accuracy setting?  If that's normal behavior I can accept that.  I just haven't seen anywhere else that outlines that.

ritchiedc, 2021-11-29:
That is correct and your right it's not spelled out clearly anywhere that I've seen.

Sebazzz, 2021-11-29:
I misread the description of this issue. Yes, the probing settings (because it is probing what the homing procedure does), use the temperatures from as set in the menu. You can set them to zero of course, then nothing will happen. Also, if you currently have higher temperatures set, those will be used instead of the set temperatures (for instance if the nozzle is at 210, that temperature would be kept).

closed 2021-11-29 by Sebazzz

talentedmrripley, 2021-11-29:
Got it.  Thanks for the clarification.