CR6Community/Marlin #290: [BUG] Cant's install firmware

tekgamerpt, 2021-12-06:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

When trying to install Community Firmware on CR6 SE with 1.1.0.3 board something goes wrong because screen red bar goes full and stays like this. Tried with several sd cards always respecting instructions for updating the firmware.
Updated the screen first and printer after but no luck. Screen goes everything ok but with printer i cant's install it.

### Version of CR6Community Firmware

6.1

### Printer model

Creality CR-6 SE

### Motherboard

Other (I will explain below)

### Display

Creality CR-6 stock touch screen

ritchiedc, 2021-12-06:
Did you also flash the screen firmware and if so did you include the kernel upgrade files?

tekgamerpt, 2021-12-06:
Yes, in fact, i flashed the screen first. Copied the Dwin folder to a formated sd card with all files and was everything ok, but even so, i flashed it 3 times just to be sure that this was not the problem.
When i went back to the stock firmware on screen the build information of the firmware installed on printer was the stock one, so something is happening that prevents flashing with the community firmware, maybe my board, i don't know.

Sebazzz, 2021-12-06:
You may need to retry several times. Creality stock boards are notorious for partially applying firmware updates. Make sure you've also applied the kernel update to the screen.

tekgamerpt, 2021-12-06:
I tried several times... 
Im not the only one with this problem.

https://damsteen.nl/blog/2021/03/27/creality-cr6-community-firmware-release-6-is-here

Honestly i dont know what can i do more. Its not my first printer.
Maybe someone can figure it out

ritchiedc, 2021-12-06:
Did you copy the 3 files (files only, not the folder) in the kernel upgrade folder to the dwin_set folder? A photo of the blue screen when the screen flash is done would be helpful.

tekgamerpt, 2021-12-06:
Those files are by default in the folder of the screen firmware, and yes flash was successful like shown on instructions, but for sure, i made it 3 times.
The problem is on the board not the screen i think

Sebazzz, 2021-12-06:
It would be interesting to see a picture of the screen PCB. From a good source I have understood the old DWIN displays that were delivered in 2020-now printers are no longer produced and Creality is now shipping new displays.

tekgamerpt, 2021-12-06:
I can provide this but not now cause its printing with the stock firmware. This printer was purchased 5 days ago. Was a promo on tomtom.com

N1c093, 2021-12-08:
I also bought the CR6-SE on tomtop on the 09. September. I successfully flashed the community firmware on the printer with board `CR-ERA_V1.1.0.3`

Is there anything I can help with?

tekgamerpt, 2021-12-08:
> I also bought the CR6-SE on tomtop on the 09. September. I successfully flashed the community firmware on the printer with board `CR-ERA_V1.1.0.3`
> 
> Is there anything I can help with?

Well, i did everything as described on instructions. If you did something diferent and worked please tell me

ritchiedc, 2021-12-08:
If you are installing 6.1 the kernel upgrade files are not in the dwin_set folder by default. They have to be copied there.

tekgamerpt, 2021-12-08:
https://imgur.com/a/r8TovEn

Here are the files on dwin folder

ritchiedc, 2021-12-08:
If those are the files in dwin_set and you did not copy the kernel files to it then you are not flashing v6.1. If you have the ERA board you need to download "CF6.1-Final-cr6-se-v4.5.3-mb-2021-06-04-16-53.zip". Is this the file you downloaded? Please be patient with me. I am trying to eliminate the most likely problem.

tekgamerpt, 2021-12-08:
> If those are the files in dwin_set and you did not copy the kernel files to it then you are not flashing v6.1. If you have the ERA board you need to download [...]

Yes, that file .zip. I did all steps like devs wrote on this github. Problem is on the board side, not the screen. The screen shows everything flashed ok like instructions. I think that it freezes on the full red bar at startup because of the stock firmware in printer.

ritchiedc, 2021-12-08:
Try a different sd card with a fresh format. Also make sure you rename the .bin file. The name has to be different from the previous filename flash attempt.

tekgamerpt, 2021-12-08:
Already did that... As i said, its not my first printer... I have a custom ender 3 V2 and i compile the firmware myself.

tekgamerpt, 2021-12-10:
Update for other users with same problem. I managed to flash the printer. Flashed with 6.0 and after with 6.1 and everything is running properly. Used same sd card as before but this way, worked.
Cheers and thanks to everyone who tried to help.

jfpk, 2021-12-16:
I am having the same problem, on a brand new printer. The display firmware flash appears to finish correctly.  When I remove the microSD card and power-cycle the printer, the splash screen appears on the display, with a red progress bar that fills up.  It wont progress past this.  I have re-flashed numerous times.

I have an ERA 1.1.0.3 motherboard on the printer, and have attached an image of the display board.

I only have one microSD card available at the moment.  But I have ordered another one to see if a different card will help.  My existing card is 64GB SDXC card, however I partitioned to 8GB and formatted FAT32/4096.  I know the instructions state SDXC cards wont work, so perhaps that is the issue.  However the display appeared to read all files correctly during the flash process.   The new card I ordered is SDHC.

I will also try flashing to the 6.0 release first, and then 6.1, as it appears that solved the issue for the original poster.

[image]
[image]

ritchiedc, 2021-12-16:
Jfpk, did you copy the kernel upgrade files (three of them) to the dwin_set folder before flashing the screen?

jfpk, 2021-12-16:
Yes I did copy those 3 files into the DWN_SET folder.

jfpk, 2021-12-18:
I have now tried it on a 2nd microSD card, with the same results.  Here is the display of the flash in progress, the file counts appear to match what is on the card, so I think its reading correctly.

[image]

Sebazzz, 2021-12-18:
What happens if you tap the Creality logo or the progress bar area?

Met vriendelijke groet,
Sebastiaan Dammann

[display_flash]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F12667515%2F146649484-1cc34037-7a3c-4457-8fa3-537b46747df6.JPG&data=04%7C01%7C%7Ce5e8217d85fd49f9b2b708d9c248279c%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637754437432491488%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=tX4CWs%2FznfWsZ%2BTLN2bGFELVyF%2FHZ6W3vBFvJp0KswQ%3D&reserved=0>

jfpk, 2021-12-18:
The screen is un-responsive to input.  I have tried tapping all over the screen, and it doesnt respond.

Sebazzz, 2021-12-18:
Try re-applying the CR-6 kernel up date files

Met vriendelijke groet,
Sebastiaan Dammann

geek111, 2021-12-28:
Have same issue :D But now know why I'm wrong. Probably you wrong too :D 
If you want upgrade firmware on printer you must put sd card on printer not screen. You put sd card on screen only if you want upgrade firmware on screen :) and probably for this you this mistake :)

RWBRITT, 2021-12-28:
I have an ender 5 pro with v4.22 board. I installed bltouch and got factory firmware to flash from SD card. I want to add a filament detector and some filament profiles. I can compile for the STM32F103RET6_creality board and it doesn't ever install completely. I can put factory bin file on card and it will load. STM32F103RET6_creality_maple is the other option I can compile for. 🤷🏻‍♂️

Djcrew6, 2022-02-11:
Update on my case.
Cant take full credit, had a big help from Davids post: https://disqus.com/by/disqus_SAhO6FGvYx/
Did as he did and it works like a charm.
So you need to upgrade the motherbord first, and then the screen.
But I notice one thing, you can´t use any SD card under 8 gig when doing the update on the motherboard 1.1.0.3, the motherbord wont notice it and then it wont upgrade. Tried alot of sizes under 8 gig with no sucsess. when I did it with a 8 gig, it workt like it shoud do.
Br Dan

jfpk, 2022-02-11:
@Djcrew6 Thanks for the info.  I had given up on the upgrade,  I will try flashing motherboard first.