CR6Community/Marlin #291: [BUG] Auto Level not working

Pray4Tre, 2021-12-13:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

Auto bed level heats up, moves to home and then just sits there with a "Preheating and Auto Leveling in Progress, Please Wait..." screen. It doesn't level or do anything. Sometimes it gets farther along and will almost do the first text point but never actually does it or completes it.

### Version of CR6Community Firmware

20210604-165244

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.2

### Display

Creality CR-6 stock touch screen

ritchiedc, 2021-12-13:
Sounds like the strain gauge system is pretriggering. Search the community firmware discord server and you'll see this is a recurring theme. You may find it is a hardware issue not a software bug.

Sebazzz, 2021-12-13:
Yes, you may want to tune your strain gauge like here: https://damsteen.nl/blog/2021/04/19/on-strain-gauge-leveling

closed 2021-12-13 by Sebazzz

Pray4Tre, 2021-12-13:
Why was this marked closed?
This has nothing to do with strain guage, it never even touches the bed.

Thinkersbluff, 2021-12-14:
I understand your thought process. The Community Software v6.1 ABL function did operate correctly on my 4.5.2 board and on my BTT board and on many other systems. Unless you are using your system in some edge-case way, which your bug report does not describe, the team is going to believe that the problem is most likely not in the firmware.  We do, however, have a lot of users who eventually report that there was actually a problem in their hardware or the configuration of their printer that “tricked” the probing system into thinking it had already hit the bed, before it tried to start the probe cycle or after that but before it hit the bed.  There are just too many variables and possible scenarios to just list you a bunch of things to try, based on your bug report.  If you would like some help to troubleshoot your system, I recommend that you search the Discord on keywords or just browse the bed-levelling channel to get a sense of what others have learned while debugging this issue on their own machines.

Pray4Tre, 2021-12-14:
It stops the second the metal thing on the gantry hits the red laser on the left. It doesnt touch the bed at all, its inches away. When the original firmware would go past once that metal piece hit the laser and then stopped once pressure was applied to the nozzle. I don't see how its a hardware issue when I had no issues on previous firmware.

I had no issues autohoming or anything prior to this firmware

Thinkersbluff, 2021-12-14:
I understand confusion when stock firmware works and Community Firmware does not, with the same hardware.
I can confirm that the Community Firmware probing routine will test for the nozzle activation state only after the optical sensor detects that blade on the gantry.  If the blue led is on at that point, the printer will act as though the nozzle has just hit the bed at that point.  I do not know how the stock firmware works under those conditions.  Maybe reverting to stock will help you determine whether this problem is indeed a new hardware condition that coincidentally struck after you flashed the CF or there is some difference between the probe routine in the CF and the stock firmware, for the same hardware configuration.