CR6Community/Marlin #292: [FR] Tell the mainboard version by the user manual

honigbrai, 2021-12-14:
### Is your feature request related to a problem? Please describe.

It helps with board identification.

### Describe the feature you want

I looked through some cr6se user manuals and saw a difference at the "Circuit Wiring"-page and noticed that it was corresponding with the mainboard.

User manual from jul 2020 https://www.creality.com/de/download:
[image]

User manual from nov 2020(Manual V1.2) https://workupload.com/file/VqGn4H6wegJ:
[image]

ajn142, 2021-12-14:
Good catch. I’m not sure how useful it is, as by the time you have the motherboard that visible you can see the silk-screened board version number. Does this change in wiring result in something that the community firmware could detect and indicate a firmware/board mismatch?

honigbrai, 2021-12-14:
I don't know, but you maybe don't need to open up your printer to see the mainboard version.

ajn142, 2021-12-14:
> version.

Wait, are you suggesting that folks just read the manual that came with the printer and assume hardware motherboard version based on whether the manual appears to have the 4.5.2 motherboard or the 4.5.3? I don’t know that I would trust the manual to have cut over to the new motherboard at the exact same time that the manufacturing line did.

honigbrai, 2021-12-24:
I think so we just need to collect data on that

Sebazzz, 2021-12-30:
I don't trust the manuals to be representative of what is inside the printer, and the consequences of having it wrong can be either not heating or actually driving the hotend into the bed, which may be a PEI sheet that is easily damaged.

closed 2021-12-30 by Sebazzz