CR6Community/Marlin #293: [BUG] Change Filament Screen - Editing Nozzle Temp Not Working

jei-gaither, 2021-12-16:
### Did you test the latest `extui` code or prerelease?

No, I cannot complete

### Bug Description

On the Change Filament Screen, changing the Nozzle Temp value in the field at the top of the screen does not change the actual nozzle temp. However, tapping on the Nozzle Temp icon/section at the bottom of the screen and entering a value there immediately cranks up the fans and adjusts the nozzle temperature.

### Bug Timeline

I just upgraded to Community Firmware CR6Comm-Rel6.1 from stock firmware. Not sure which version this started in.

### Expected behavior

I expected that tapping on the Nozzle Temp display field at the top of the screen, and then being prompted to enter a new temperature would adjust the nozzle temperature, however nothing happened until I tapped on the Nozzle temp icon/temperature at the bottom of the screen and then entered a value in (presumably) the same overlaid entry keypad.

### Actual behavior

Nothing happened when the keypad overlay was displayed after tapping on the top Nozzle temperature field and a new value was entered.

### Steps to Reproduce

1. Go to PREPARE screen, then tap on the "Nozzle Temp." field at the top of the screen.
2. Enter a value in the overlaid keypad.
3. Hit the Green Checkmark button.
4. Nothing happens.
5. Tap on the Nozzle icon/temp icon at the bottom left of the screen.
6. Enter a value in the overlaid keypad.
7. Hit the Green Checkmark button.
8. Fans/heatup process immediately begins.

### Version of CR6Community Firmware

CR6Comm-Rel6.1

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.2

### Display

Creality CR-6 stock touch screen

### Mods

None

### Add-ons

None

### Your Slicer

Prusa Slicer

pollinolas, 2021-12-19:
you have to press the 'load' or 'unload' button to set the selected temperature

Thinkersbluff, 2021-12-19:
I followed steps 1-4.  Instead of "nothing happens", the value in step 1 changed to the new value.
If that is not happening on your system, there is something mysterious wrong on your system (e.g. the mapping of the Red X button on the overlay screen may be underneath the Green Checkmark) but that is not by design.  Re-flashing may help.

If you expected the nozzle to start heating to that temperature, then no, as @pollinolas says, you need to initiate either a load or an unload, before the nozzle will actually start heating.  NOTE: If you try tapping that nozzle temperature at the bottom of the screen or editing the target temperature in the box AFTER initiating either a Load or Unload, neither will work.  At that point, you are "locked" into the target temperaure until the end of the process, unless you cycle printer power.

If you tap on any of those four parameters displayed at the bottom of any screen, you are likely to find they can be modified. It is a kind of "easter egg", in the absence of a user manual; an optional short-cut convenience.  It will not, however, result in a filament change.