CR6Community/Marlin #294: [BUG] It sets temperature to 0 when leveling

JohnnyWobble, 2021-12-24:
### Did you test the latest `extui` code or prerelease?

No, there is no prerelease

### Bug Description

Every time it goes down to probe the bed slowly, whether during ABL or just a G28, for a brief second, it sets the temperature to 0. Normally this wouldn't be a problem, but because it's using an `I` gain in the PID controller (I'm fairly sure this is the problem), the temperature will actually drop about 15 degrees while it builds up enough error to achieve the desired temperature.

### Bug Timeline

I'm using 6.1, I haven't used community firmware before this version

### Expected behavior

I would expect it not to drop the setpoint temperature to 0

### Actual behavior

As it moves down to home the second (slower) time, it sets the temperature to 0, then restores the temperature once it goes back up, but this causes a large drop that delays everything by like a minute.

### Steps to Reproduce

1. Heat up nozzle (mine is at 220, but its happened at any temperature)
2. Select "home" in the movement menu (or give it a file with g28, or ABL it)
3. On the second probe, watch the temperature, it will drop

### Version of CR6Community Firmware

CR6Comm-Rel6.1

### Printer model

Creality CR-6 SE

### Motherboard

Creality v1.1.0.0-ERA

### Display

Creality CR-6 stock touch screen

### Mods

Ender 3 v2 Extruder setup

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

ritchiedc, 2021-12-24:
This is per design. Refer to the release 6.1 FAQs in the release notes which mentions how to disable that function.

JohnnyWobble, 2021-12-24:
Fascinating, thank you!!

closed 2021-12-24 by JohnnyWobble