CR6Community/Marlin #295: [BUG] Resume after Pause at height (M0) not working

DavHau, 2021-12-25:
### Did you test the latest `extui` code or prerelease?

No, there is no prerelease

### Bug Description

I used the pause at layer script in Cura 4.9 to change the filament to another color.
It uses the M0 command to pause.

After changing the filament it was not possible to resume printing after the pause.
The display still shows the pause button, despite the print already being on pause. 
Since there is no resume button, it is not possible to resume the print.
Even clicking the pause button again does not change it into a `resume` button.

related:
  - https://github.com/CR6Community/Marlin/issues/17
  - https://github.com/CR6Community/Marlin/issues/234

### Version of CR6Community Firmware

Rel6.1

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.3

### Display

Creality CR-6 stock touch screen

### Your Slicer

Cura

Sebazzz, 2021-12-30:
Do you have a minimally reproducible file?

It must be noted that other methods of pausing work better, because M0 is essentially an emergency command. Still, it should work as intended.

DavHau, 2021-12-30:
Here the gcode. Roughly a 1.5h print. Probably 1h until the pause

[heart_with_pause_for_filament_change.gcode.zip](https://github.com/CR6Community/Marlin/files/7794716/heart_with_pause_for_filament_change.gcode.zip)

Thanks for the hint, I will try with another pause code next time.
I'm wondering why the cura script recommends M0 for the Marlin firmware if it is not the right code in practice.

Sebazzz, 2021-12-30:
A _minimally reproducible file_ that reproduces the issue please (preferably without or minimal extrusion) :)