CR6Community/Marlin #296: [FR] Make Firmware Retraction Settings Available from the SETUP menu

ritchiedc, 2022-01-05:
### Is your feature request related to a problem? Please describe.

The firmware retraction settings are only available after starting a print.

### Describe the feature you want

It would be convenient to be able to change the firmware retraction settings (distance, speed, etc) before starting a print.

Sebazzz, 2022-01-14:
That doesn't work in this fork because the retractions are reset on every print.

It is in this firmware to be used through either:

- Temporarily overriding settings
- Emitting the FW retract settings in the start or filament gcode like this: https://damsteen.nl/blog/2021/05/06/adjustable-retractions-during-printing-with-marlin-firmware-retractions - different prints need different retraction settings

ritchiedc, 2022-01-14:
It saves the changes to the firmware retraction settings when settings are saved. To test I started a print and made a change to the retraction length, then made a change to the z-offset which forced the settings save. I allowed the print to complete, cycled power to the printer and when I started the print the retraction length was still at the new value. Seems this would work when not printing as well.   To test this I issued an M207 with a length change followed by an M500. I then cycled power and the M207 reported the new value.

Am I missing something in your explanation?

Sebazzz, 2022-01-14:
Yes, it resets once you start a new print 😉

ritchiedc, 2022-01-14:
The override resets but not the retraction settings. I have firmware retraction enable in both Prusaslicer and Cura. The G10 and G11 commands still use these settings.

Sebazzz, 2022-01-14:
Of course, you’re right 😊

Yes, it could be added to the set-up menu then – but we’d need to think about this on how not to make it confusing.

ritchiedc, 2022-01-14:
All I ask is that it be considered. For my use case it would add convenience.