CR6Community/Marlin #297: [BUG] Bed Leveling temp issues

Luk3du, 2022-01-24:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

Whenever doing ABL from the menu option, the temp sets correctly but shows as "0" when the z axis is moving and then goes back to the set temp after, this happens for every probe point 

### Bug Timeline

I did not notice this happening on stock fw, only on 6.1 community 

### Expected behavior

Temp stays on at all times 

### Actual behavior

Temp sets to 0 on z move during abl 

### Steps to Reproduce

"run automatic bed leveling" from leveling menu 

### Version of CR6Community Firmware

6.1 

### Printer model

Creality CR-6 SE

### Motherboard

Creality v1.1.0.0-ERA

### Display

Creality CR-6 stock touch screen

### Mods

No

### Add-ons

N/a

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

Thinkersbluff, 2022-01-24:
Have you read the readme file?

Luk3du, 2022-01-24:
> Have you read the readme file?

Yes I read the readme files included in the 6.1 folder and the one on GitHub, haven't seen anything mentioning this anywhere?

ritchiedc, 2022-01-24:
Just copied this from the faqs

"The temperatures go up/down/up/down while bed leveling

That is by design to improve bed leveling accuracy. You can turn that behaviour off by disabling "Better accuracy" in the leveling settings (not recommended)."

Sebazzz, 2022-01-24:
Yup, it is mentioned and can be disabled as desired.

closed 2022-01-24 by Sebazzz