CR6Community/Marlin #298: [FR] When printing with no bed temperature, disable warming up for homing

gearlappen, 2022-01-27:
### Describe the feature you want

When printing TPU, I want the bed to stay cool so I let the slicer write 0 into the GCODE. The CR6SE CF 6.1 homing procedure heats the bed up. After that it lets the bed cool forever. 

My proposal is to implement a check like this:

if the target temperature is 0 -> don't heat up for homing

Thinkersbluff, 2022-01-27:
Have you tried setting the bed temperature  pre-heat setting to zero?  
[image]

gearlappen, 2022-01-27:
No, but this would be a work around for this particular case of course.

ritchiedc, 2022-01-27:
Your solution would require a work around for my use case where I don't set temperature until after homing. It's likely it would affect a larger number of people since this method is used by Sebazzz's recommended start gcode. There's always trade-offs.