CR6Community/Marlin #299: [FR] Tune>Override>Slicer: Keep fan on/off, Set permanent temperature

mbourque, 2022-01-27:
### Is your feature request related to a problem? Please describe.

Many times I wish to print the same gcode file and but want to ignore temp, fan changes with an override setting just like Slicer Retract Override.

### Are you looking for hardware support?

no

### Describe the feature you want

I would like to enhance the Slicer Override features to include:

- Override fan on/off/speed changes in slicer (so Marlin will ignore all fan speed and on/off)
- Override nozzle temperature changes in slicer (so Marlin will ignore any temperature changes)
- Override bed temperature changes in slicer (so Marlin will ignore any temperature changes)

Sebazzz, 2022-01-27:
I suggest you request this feature at upstream Marlin first. The override slicer retractions feature resells Marlin's undersold firmware retraction feature and builds an usable interface on top of that. 

But these changes have quite some functional impact. While I'm happy to try and implement this, upstream they may have better ideas what impact this has. The fan speed and nozzle/bed would also mean you couldn't control your printer from Octoprint anymore. It couldn't distinguish what you are printing and what you are commanding.

mbourque, 2022-01-27:
Hmm those are good arguments I hadn't thought of. Let me think about it more and perhaps post to Marlin upstream. Thank you for your thoughtful response.