CR6Community/Marlin #309: [BUG] M907 Current control not enabled on stock CR-6 Max firmware (untested with other builds)

printd3v, 2022-02-08:
### Did you test the latest `extui` code or prerelease?

No, I cannot complete

### Bug Description

M907 control is disabled on stock board, even though steppers user trinamic 2209 drivers, (which have this enabled). Can confirm stock firmware has this enabled, but obviously can't change because no saving to EEPROM.

### Expected behavior

M907 control for current on motors.

### Actual behavior

M907 is not enabled in marlin

### Steps to Reproduce

Update Marlin configuration.adv to have M907 enabled.

### Version of CR6Community Firmware

Release 6 (Final)

### Printer model

Creality CR-6 MAX

### Motherboard

Creality v4.5.3

### Display

Creality CR-6 stock touch screen

### Mods

Orbiter v1.5

ritchiedc, 2022-02-09:
You cannot control motor current from the software using the stock creality board, it's a hardware design limitation (my understanding is that they didn't include the serial interface to the drivers). You can only control the current using the vref pots on the board. If you want to control the current from firmware you have to use the btt skr cr6 board.

printd3v, 2022-02-09:
No, in multiple iterations of the cr6 max firmware, there are changes to the UART control. UART is what communicates with the drivers on the board. I noticed minor tweaks and changes, and I ended up making the change for my orbiter extruder. I took a look in the firmware, and the values are set by UART, meaning that it is firmware-based.  I noticed a change from one motor from 600 to 550 mA. I don't see a reason why M907 shouldn't be enabled. Creality seemingly also did have this enabled to, probably as they were doing development with it, for ease of tuning in pronterface or any other similar program.

printd3v, 2022-02-09:
I’ll end up calculating the vRef tomorrow, and measuring the board with a potentiometer just in case. If this is the case, why would creality even have it enabled in the stock firmware?

closed 2022-02-09 by printd3v

Sebazzz, 2022-02-10:
The background of that Creality boards don't support UART is because "It is slow" (literal words from a Creality engineer who I asked about it)