CR6Community/Marlin #310: Nozzle refuses to re-heat after initial time out

mputtr, 2022-03-12:
### Did you test the latest `extui` code or prerelease?

No, there is no prerelease

### Bug Description

I have been doing a series of filament change prints this past week or so and I've noticed that during a filament change command, if I wait too long, the nozzle would time out (as it should), but after changing the filament, any attempt to change the temperature, either by changing it from the nozzle icon on the bottom of the screen, via the Tune screen, or via the filament change screen does not trigger a reheat of the nozzle and instead triggers a thermal error.

The only way to recover is to test my luck and turn the printer on/off and hope the print resume function works. If it works, then it will start to heat up the nozzle again, if the print resume function doesn't trigger, then I have re-slice from where I last left off and glue the pieces in. 

I have tried turning the power on/off quickly (within 1 second) and also by giving it a longer interval (5+ seconds & 5 mins). I was unsucessful and triggering the print resume function.

### Bug Timeline

I don't know for sure, but I know I see this in v6.1

### Expected behavior

Nozzle should re-heat back to the temperature selected after time out. 

### Actual behavior

nozzle doesn't re-heat.
Thermal Error page triggered requiring a printer re-start

### Steps to Reproduce

wait for filament change command
wait for printer to time out on maintaining nozzle temp
change nozzle temperature either from the bottom of the screen, or via Tune/Nozzle Temp, or via, Tune/Other Settings/Change Filament/Load filament
wait for about a minute
Thermal Error requiring a restart

### Version of CR6Community Firmware

6.1

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.2

### Display

Creality CR-6 stock touch screen

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

SD Card (headless)

Sebazzz, 2022-03-14:
I suggest you PID tune.

mputtr, 2022-03-14:
> I suggest you PID tune.

I used the auto PID tune when I did the initial install of 6.1 
I did a factory reset then used the auto PID tune function.

I am starting to think that maybe the main issue is that the manual filament change function is not triggering the heating up, while the resume function will heat the nozzle up.

jonyo, 2022-03-19:
Same issue, though for me the filament sensor is triggering the change filament on a slow part of the print (guess I need to adjust that to not be as sensitive).  But once the heat timeout is reached, no matter what it refuses to heat back up while it is paused on "filament change".

After reading this thread, I did the PID tune in the menu, it did some stuff then said complete.  Still same problem on the next print.

dphilipov, 2023-11-04:
I have the exact same problem. The only way to get the nozzle to heat back up again is to select `Resume Print`. However, if the temp starts dropping during a filament change sequence it get's stuck there forever.

hotdog374657, 2023-12-04:
I'm having the same issue when trying to print Hueforge prints