CR6Community/Marlin #311: Help- V6.1 Firmware flashed incorrectly

3FMGreg, 2022-03-20:
### Description

Unable to correctly flash firmware V6.1

### Steps to Reproduce

See attached file "Issue - installing Community FW V6.1.pdf"

### Additional Information

CR-6 SE. Motherboard v4.5.2, no mods

[Which *exact* firmware version did you flash? Provide a link to the download you have used.]
https://github.com/CR6Community/Marlin/releases/download/v2.0.8.1-cr6-community-release-6.1/CF6.1-Final-cr6-max-stock-mb-2021-06-04-16-52.zip
[Issue - Installing Community FW V6.1.pdf]

Ron1776, 2022-08-06:
I have a Creality CR-6 SE with CR-ERA_V1.1.0.3 motherboard which I understand is the latest v4.5.3.

https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-5-beta/CF5Beta-cr6-se-v4.5.3-mb-2020-12-30-14-19.zip

I successfully flashed the touch screen but the motherboard flash results in a Community edition start screen with the progress bar blank (0% loaded).

ritchiedc, 2022-08-06:
First to make sure we're talking about the same thing, the ERA motherboard is compatible with the v4.5.3 motherboard. Second, the latest release is 6.1 final. Finally, you need to make sure the screen DGUS os version is 4.1 or less to use the screen firmware that comes with that release. You may want to go to the cr6 community discord server to get more help.

Thinkersbluff, 2022-08-07:
@Ron1776 - if you are having trouble getting the firmware to load, that is not necessarily a problem with the firmware.
The Creality boards need you to rename the firmware file, each time you try to flash it (it will ignore a bin file if it recognizes the filename from the previous flash)
They also need you to flash with an SD card formatted FAT32, 4096 sectors.
The 0% progress bar may be a sign that the last MB flash did not work.

These FAQs may help: https://github.com/CR6Community/Marlin/wiki/How-To-Flash-Firmware-to-the-CR6-printer