CR6Community/Marlin #312: [FR] Auto increase caliration mesh height mid print at specific location when the nozzle touches the bed/print on the same spot x amount of times

ANTONIOPSD, 2022-04-10:
### Is your feature request related to a problem? Please describe.

When I want to print something important I always run an auto level calibration but when I startr printing I can see the blue light on the pressure sensor turning on on some areas and that means that the nozzle is touching/scratching at that location.

Could be possible to add some feature to auto increase a little bit the offset on those locations so the layer height remains constant? This could be called "Mid Print Auto Calibration"

This will dynamically adjust the calibration on each print.

### Describe the feature you want

Start printing
Listen for pressure sensor activations
When the pressure sensor activates, get nozzle location and add 1 to that area or area with offset
If nozzle touches the same spot on next pass, increase calibration mesh height on that spot by 0.xx mm

I don't know  if:
1. Pressure sensor can be accessed mid print.
2. Calibration mesh can be modified/reloaded mid print.

Thinkersbluff, 2022-04-10:
The blue led will also flash if you tap the Bowden tube or snag the filament or strike a blob on the print. It is not a reliable indication of ABL mesh errors.  I think most of us see that light flicker a little, even as the head is transiting from place to place above the bed.  If it is flickering too often, the probe gauge may be under strain from something like tension on the heater wire harness or the activation force may be set too low.

ANTONIOPSD, 2022-04-10:
> The blue led will also flash if you tap the Bowden tube or snag the filament or strike a blob on the print. It is not a reliable indication of ABL mesh errors. I think most of us see that light [...]

Yeah, I thought that as well, that's why it should only happen when the detections happens on the same X/Y location for x amount of times because accidentally tapping or snagging happens randomly and should be easy to ignore and about the blobs.. well you are right but it could be also ignored if the force value could be accessed.

I only know python and I know nothing about marling firmware but I could read and understand the code and try to implement it by modifying the leveling mesh on the fly just to test how it woks but I don't know wich files contain the code related to that.

Thinkersbluff, 2022-04-10:
The goal of this Community Firmware project is to facilitate the merging of this firmware back to the upstream Marlin. I think what you propose would require a unique deviation from that main branch, but you might find if you post your issue upstream you can get better help.

closed 2023-03-28 by ANTONIOPSD