CR6Community/Marlin #315: [BUG ?] touch input on the screen does not work

junonara, 2022-04-18:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

I bought a new CR-6 SE about 2 weeks ago. I tested printing without upgrading the firmware for about a week, and recently upgraded the CR6SE Community firmware (HW + Touch Screen). After upgrade, touch input doesn't work.

Board : ERA 1.1.0.3
LCD : Stock touch screen
[image]

Original Firmare version : 2.0.1.8
Firmware used for upgrade: CF6.1-Final-cr6-se-v4.5.3-mb-2021-06-04-16-53.zip

The SD card used is a Sandisk 8G, formatted as FAT32, 4096. I copied 3 kernel files into the DWIN_SET folder. Touch screen shows END indication and it has been completed normally. After removing the SD card and booting, the boot screen is displayed, the main screen is displayed, and the icons are displayed normally. But when I click on the icons, nothing happens. By connecting Octoprint, the printing output proceeds normally. Therefore, printer HW has been upgraded normally and it seems to work normally.

I re-uploaded the official  2.0.1.8 version firmware(HW + Touch screen) of Creality, but the touch action still doesn't work. I am not sure if this symptom is due to HW or firmware.

What's particularly odd is that the LCD dims a bit if left unattended. It's probably a screen saver, but if I touch it when it's dimmed, it's bright again. The touch also seems to work partly, but when I touch the icon, nothing changes.

### Expected behavior

I wish it could be a touch input

### Version of CR6Community Firmware

CF6.1-Final-cr6-se-v4.5.3-mb-2021-06-04-16-53

### Printer model

Creality CR-6 SE

### Motherboard

Creality v1.1.0.0-ERA

### Display

Creality CR-6 stock touch screen

Thinkersbluff, 2022-04-18:
Hello!  Sorry to hear of the troubles you are having.  This is the first I have heard of the touch screen becoming non-responsive to the menu buttons, after flashing.  It is doubly-concerning that you have also not been able to revert to the stock firmware.

I can try to help, but I do not know enough to make any assurances.  If you are willing to experiment and to share your experience here, then maybe we can help both you and the next folks who encounter this trouble.

First, please confirm that your main board is actually ERA v1.1.0.3, not v1.1.0.0, as entered above.  We have no knowledge of a 1.1.0.0 board.

Yes, the Community Firmware presently requires DGUS2 v3.5 to be the Display OS, which is why flashing the kernel upgrade files is a necessary part of flashing the display.

If Creality is not including the DGUS kernel files in their firmware distribution files, that is going to leave you with v3.5 as the display OS when you revert to stock.

I do not know whether the Creality 2.1.0.8 firmware runs on DGUS2 v3.5.  If not, that could explain your current issues.
Do you remember what DGUS version was installed, before you flashed v3.5?

It may be prudent for you to ask Creality themselves (or your supplier, if you did not buy directly from Creality) for the correct DGUS kernel files. ( You may not need to explain why you are asking)

If it was DGUS2 v4.0 before, it is possible that adding this file to the Creality DWIN_SET folder and reflashing their stock firmware to your display will resolve the reversion problem, but I cannot say for sure, since I have different hardware than you.…: https://www.DWIN-global.com/uploads/TL5_UI_DGSU2_v40_Fast-boot_20210917.zip
( and yes there is a typo in that URL, but it works when entered that way)

I have scanned that download with Norton360 and found no viruses in it.  I have successfully flashed it to my own CR6-SE printer.   Using a new pre-release version of the CF UI, which is still under development, I have also been able to upgrade my display to DGUS2-v4.0Fast with that download and run CF6.1 on it.

If you are willing to try the above download, please let me know here what was the result.  If you get other info or help from Creality on this issue, please share what you learn.  I will update the CF FAQs with what you can confirm has worked.

junonara, 2022-04-19:
My CR-6 SE board is ERA-1.1.0.3. Sorry for typing it wrong.

I remember the original kernel version of Touch Screen as "T5L1 DGUS2 V4.5 2022.01.13".

I sent a kernel request mail to the Creality CS team. I've looked for V4.5 related kernels on the DGUS site, but only the latest version (4.7) and I'm not sure which file to choose exactly.

There are three kernel files for community touch screen firmware : T5L_OS_DGUS2_V10(no RX8130), T5L_OS_V11_RTC(RX8130), T5L_UI_DGUS2_V35_20200730. Are all these files necessary to change the kernel?

Thinkersbluff, 2022-04-19:
> I sent a kernel request mail to the Creality CS team. I've looked for V4.5 related kernels on the DGUS site, but only the latest version (4.7) and I'm not sure which file to choose exactly.

Ok!  I have learned a lot today!
On the Community Discord, we have a member who also needed to flash their system back to v4.5.  They have posted a copy of the bin file with a discussion of the issue here: https://discord.com/channels/759603746270609428/804078252661800981/965786740084342874

junonara, 2022-04-19:
Thank you.

After installing the Creality original firmware version 2.0.1.8 on the CR-6 SE and installing the Touch Screen Kernel T5L_UI_DGUS2_V45_20220105 version, I confirmed that the Touch input works.

Thinkersbluff, 2022-04-19:
> After installing the Creality original firmware version 2.0.1.8 on the CR-6 SE and installing the Touch Screen Kernel T5L_UI_DGUS2_V45_20220105 version, I confirmed that the Touch input works.

Excellent!

So now we have confirmed that the current CF6.1 UI firmware does not run on any version of DGUS2 except v3.5, and v3.5 apparently does not work on your display hardware.  

There is a new version of the UI firmware under development, which has been redone specifically in anticipation of this particular evolution in the DWIN displays.  It is now almost ready for user testing.  Would you like to be a part of that testing?

Thinkersbluff, 2022-04-20:
It would be interesting and helpful to know whether this DRAFT new version of the UI works on your system.

https://discord.com/channels/759603746270609428/804078252661800981/966203256520982538

junonara, 2022-04-21:
I tested CR-6 SE with the DRAFT version.

1. CF6.1-Final-cr6-se-v4.5.3-mb-2021-06-04-16-53 + DRAFT DWIN_SET + DRAFT V40 Fastboot kernel : Screen is visible but touch input does not work
2. CF6.1-Final-cr6-se-v4.5.3-mb-2021-06-04-16-53 + DRAFT DWIN_SET + DRAFT V45 kernel : Screen is visible and touch input works

Thanks

Thinkersbluff, 2022-04-21:
Thanks for the feedback!
Funny, I get exactly the opposite response from my own (Kickstarter) CR6-SE.  Mine works with 4.0 and not with 4.5.

Thanks also for the photos of what is not working on the Printing screen.
Looks like we have a little more work to do, getting the DRAFT version operational. 

May I close this issue report, now?

junonara, 2022-04-21:
Yes, you can close it. I hope your work goes well.

Thinkersbluff, 2022-04-21:
I believe this version will display the Printing, Printing Paused and Printing Finished screens correctly, now.

[NextGenCF6.1DisplayOnDGUS2v4.0_DRAFT2.zip](https://github.com/CR6Community/Marlin/files/8528416/NextGenCF6.1DisplayOnDGUS2v4.0_DRAFT2.zip)

Thanks again for alerting us to your issue and to providing feedback on our ongoing efforts to address it.
I do hope you are also now a member of the Community Discord.  Happy printing!

closed 2022-04-21 by Thinkersbluff

Thinkersbluff, 2022-05-03:
The refactored DWIN_SET has now been released on the CR6Community Touchscreen Firmware Repository.

https://youtu.be/otaNJFLcuQ8