CR6Community/Marlin #316: [FR] Invert feeder stepper motor direccion from the menu

ANTONIOPSD, 2022-05-03:
### Is your feature request related to a problem? Please describe.

I changed the feeder and now the filament gets pulled from the opposite direction of the gear si it extrudes backwards.

### Describe the feature you want

Just an option in the menu to invert the extrusion direction.

rmeissn, 2023-01-30:
You can inverse the stepper motor directiong by changing its wiring: https://printingit3d.com/how-to-reverse-the-direction-of-an-extruder-motor/
That might be much easier than implementing this in the firmware (if possible at all).

ANTONIOPSD, 2023-02-02:
> You can inverse the stepper motor directiong by changing its wiring: https://printingit3d.com/how-to-reverse-the-direction-of-an-extruder-motor/ That might be much easier than implementing this in the firmware (if possible at all).

That's what I did, but couldn't this be implemented by multiplying the extrusion by -1?
It can be done by editing the G-code and multiplying the extrusion by -1, so it should be easy to implement.

I could try to add the code, but I don't know where the code that controls the extrussion is.