CR6Community/Marlin #317: Touch screen calibration wiped after flashing latest firmware

goranthales, 2022-05-11:
### Did you test the latest `extui` code or prerelease?

No, I cannot complete

### Bug Description

I recently flashed the latest community firmware for my 1.1 ERA board CR6SE (I physically checked the board to be sure), and after completing flashing, the printer will boot up correclty, but the screen doesn't register any touches.

Inserting an sd card switches to the print file select screen, and removing it goes back to the home screen and says "media removed"

If I wait the ~30 or so seconds for the screen to dim, touching it will brighten it up again. But aside from that, no touches register and I cannot do anything on the printer.

From what I've seen online, it seems the touch screen calibration has been wiped. Is there anyway to recalibrate from this state?

### Bug Timeline

Happed an hour after assembling brand new printer, only after flashing community firmware

### Version of CR6Community Firmware

CR6Comm-CF6-Final-cr6-se-v4.5.3-mb-2021-03-27-15-53.zip

### Printer model

Creality CR-6 SE

### Motherboard

Creality v1.1.0.0-ERA

### Display

Creality CR-6 stock touch screen

Thinkersbluff, 2022-05-11:
This sounds like the issue reported in closed issue #315 
The first two videos of this playlist explain the cause and the recommended fix.

Early feedback on the fix suggests that some users may seem white noise screen at the end of flashing the new DWIN_SET, which can be resolved by deleting the CFG file and reflashing.

diegofdezj, 2022-06-24:
Hi! I have the same problem, my touch display is not responding I've already installed the kernel V3.5 and V4.5 and not works I dont remember which kernel my CR-6 SE had installed and I bought a new display but not works too... can u provide me the links for download te V6 and V6.1 of the community firmware please.
I hope resolve this problem... thanks

Thinkersbluff, 2022-06-24:
Sorry, no. When you said you had "the same problem ", I foolishly assumed you had the same board, also.

Did you watch the video I posted above?  The Step 1 video tells you how to find the right file to download. 

https://github.com/CR6Community/Marlin/releases/download/v2.0.8.1-cr6-community-release-6.1/CF6.1-Final-cr6-se-v4.5.2-mb-2021-06-04-16-52.zip

Thinkersbluff, 2022-06-30:
BTW I have now posted a video demonstrating how to calibrate your screen to be compatible with either DGUS2 v3.5 or v4.5:

https://youtu.be/0xwFGiyg4Z8

To use the Touchscreen firmware bundled with the CF6.1 , you can follow those instructions to calibrate your screen to be compatible with DGUS2 v3.5.

Alternatively, you can reflash DGUS2 v4.5 and flas the refactored touchscreen firmware.

closed 2022-06-30 by Thinkersbluff