CR6Community/Marlin #318: [BUG] Resume popup at power-up after complete print

guivo, 2022-05-17:
### Did you test the latest `extui` code or prerelease?

No, there is no prerelease

### Bug Description

Most of the time, when a power-up the CR-6 the pop-up associated with the power resume appears. In all those case I have previously successfully finished the job, touched "exit" on pop-up that appear at the end of the print and powered down the machine.

### Expected behavior

Normal display homepage

### Actual behavior

Resume pop-up appears on the display, on top of the home.

### Steps to Reproduce

1. print
2. wait to end the task, including the final movements
3. touch exit
4. power down the machine with the PS switch
5. switch on the machine

### Version of CR6Community Firmware

Latest releases for both

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.2

### Display

Creality CR-6 stock touch screen

ritchiedc, 2022-05-17:
Check to see if the sd card still has the power loss recovery file on it (plr extent). Either you're not waiting long enough before powering down the printer for it to be deleted or the sd card is corrupt and it can't be deleted. If you don't need the power loss recovery feature it can be disabled from the screen.

guivo, 2022-05-17:
Thanks, I will check after the next print. I waited enough 99.9% of the times, the plate was almost cold. The problem appeared with the community firmware, I am having it since months but it is not a big issues and I didn't bother to report it.

guivo, 2022-05-24:
After I checked and repaired the filesystem the problem didn't appear again.

closed 2022-05-24 by guivo

Thinkersbluff, 2022-05-24:
> After I checked and repaired the filesystem the problem didn't appear again.

That sounds good.  Maybe the main board could see traces of a deleted file in the boot record…. Ghosts in the machine are the trickiest to debug.  We so take these cards for granted, but they have some nasty failure modes.

ok to close this?