CR6Community/Marlin #321: Compilation error

pacocatech, 2022-07-12:
I'm trying to generate a version of your firmware, however, when running the build I'm getting this error message:
"
*** [.pio\build\STM32F103RET6_creality\src\src\module
\probe.cpp.o] Error 1

"

CR6-SE board 4.5.3

Another question, I'm trying to upload a version of marlin without updating my printer's display, however, every time I try to do that I can't install the current marlin on the machine, it freezes on the loading screen, you know what Can be?

pacocatech, 2022-07-12:
My configuration and adv files:
[Marlin.zip](https://github.com/CR6Community/Marlin/files/9088364/Marlin.zip)

pacocatech, 2022-07-12:
[image]
this is the situation when i try to put marlin 2.1.x on my cr6 itself, does anyone know why this happens?

Thinkersbluff, 2022-07-12:
The Creality display is not programmed to work with Marlin 2.1, and Marlin 2.1 is not programmed to work with the Creality display.

If you want to compile and load the Community Firmware onto your system, you must program both the display and the motherboard.  They are a matched set. 

The Community Firmware source is on our GitHub page, along with the Release files.

Thinkersbluff, 2022-07-12:
Disable the Probe tare buzzer line, to fix your compile error.
[image]

pacocatech, 2022-07-12:
Thanks, disabling that line made the build work.

Is there a version of marlin I can put that I don't have to mess with the display?
I just wanted to enable the m600 command on my CR6-SE

Thinkersbluff, 2022-07-12:
None that I know of.

closed 2022-07-12 by Thinkersbluff