CR6Community/Marlin #322: [FR] Add tools to the printer through the UI

WriteCodeEveryday, 2022-07-15:
### Is your feature request related to a problem? Please describe.

Yes, when trying to print multiple colors using 3D Chameleon, Octoprint is having serious issues switching tools.

```
Invalid tool reported
Your printer reported tool T1 as invalid, reverting back to T0
```

### Are you looking for hardware support?

No hardware support needed here.

### Describe the feature you want

Somewhere under the misc settings, there would be a widget just like the Z axis offset which would allow you to go from 1 to N number of tools on your printer. It is fine if after this change is made the printer restarts.

WriteCodeEveryday, 2022-07-15:
I can exclude these from octoprint by using the "Blocked Commands" but I would like a native solution if possible.

Thinkersbluff, 2022-07-15:
Sorry - are you asking for the CF6.1 to be enhanced to work with Chameleon hardware?