CR6Community/Marlin #323: XYZ-Axes Offset Control and Build Volume Resizing

R0boT4ter, 2022-07-24:
### Is your feature request related to a problem? Please describe.

I have a direct drive extruder and I need to reduce the X-Axis down to 375mm and Y-Axis down to 365mm. The ability to change these via touchscreen is preferred as I have no idea how to decompile/recompile the bin file so I can edit the "Configuration.h" file. I am a complete novice to hexcode editing so that's not an option for me either at this time. If this can't be a feature that's fine but if someone could help with this problem in any way I would much appreciate it.

### Are you looking for hardware support?

-None-

### Describe the feature you want

The ability to change the size and offsets of the XYZ-axes via touchscreen. Both being highly preferred

Thinkersbluff, 2022-07-24:
What printer are you using?

Have you tried just modifying the build volume in your slicer, so that it will not let you place prints on the untraceable parts of the bed?

You may find M206 and M666 allow you to add offsets to the min and max end stop values.

R0boT4ter, 2022-07-26:
The ABL on a CR6 Max is the only time I have problems with collisions

R0boT4ter, 2022-07-26:
I'm currently designing a hotend and vwheel assembly that should solve my problem... but the ability to change offsets on the touchscreen would be nice