CR6Community/Marlin #325: [BUG] load/unload from screen does not operate properly

lifendel, 2022-09-06:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

If I download and compile the latest firmware from the github, if I select load, the printer heats to 200°, moves the gantry up and then right back down without moving the extruder. If I select unload, it heats to 200°, moves the gantry up, unloads the filament maybe 10mm, then immediately feeds 10+mm as it starts extruding and then the extruder skips as it is extruding too fast (it's only brief, so maybe 15-20mm) stops and lowers the gantry back down. If I download and compile the 6.1 tagged source and install that, load/unload works properly. This happens on both the SE and Max.

### Version of CR6Community Firmware

6.2

### Printer model

Creality CR-6 MAX

### Motherboard

Creality v4.5.3

### Display

Creality CR-6 stock touch screen

G00364778, 2023-03-01:
I have the same issue.

Version of CR6Community Firmware: 6.2
Printer model: Creality CR-6 SE
Motherboard: Creality v4.5.3
Display: Creality CR-6 stock touch screen

- Filament load/unload does not move at all. 
- Extrusion calibration does not move either

Thinkersbluff, 2023-03-02:
There is no formal release 6.2 of the Community Firmware.  Yes, if you download the latest Code from the repo, it has this issue. The last stable release was/is 6.1.