CR6Community/Marlin #329: [BUG] Printer restart after four touches on home screen.

OndrejH6789, 2022-10-12:
### Did you test the latest `extui` code or prerelease?

No, I cannot complete

### Bug Description

Printer will restart and does not properly power up after four touches on the LCD if it is tapped outside of control buttons on home screen.

### Steps to Reproduce

1. Start the printer
2. Touch 4 times outside of control buttons on the home screen
3. Printer do restart but does not boot up

### Version of CR6Community Firmware

CR6Comm-Rel6.1

### Printer model

Creality CR-6 MAX

### Motherboard

Creality v4.5.3

### Display

Creality CR-6 stock touch screen

ritchiedc, 2022-10-12:
This is expected behavior. If you press the logo again on the boot screen it will take you back to the main menu with a message about the devs main menu. I'm not sure what it enables but was there for the benefit of the devs.

OndrejH6789, 2022-10-12:
Thanks for clarification. I didn't know that. I tried it and it works like you wrote. Thanks.

closed 2022-10-12 by OndrejH6789