CR6Community/Marlin #330: [BUG](Print origin set at homing point)

Dennis54, 2022-10-31:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

Firmware update to Rel 5, reinstalled Cura

When starting a print after going through the tedious setup, the printer starts at the Homing point and not the origin(centre of the build plate).

### Bug Timeline

This issue occurred after error PID thermal control

### Expected behavior

only the top right quadrant of the job printed

### Version of CR6Community Firmware

Rel 5

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.3

### Display

Creality CR-6 stock touch screen

### Mods

none

### Add-ons

standard

### Your Slicer

Cura

### Host Software

SD Card (headless)

Thinkersbluff, 2022-11-01:
Sorry, Dennis, I am confused. 

-  The latest Community Firmware release is 6.1.
- The origin (X=0,Y=0) point is the front left corner of the bed, not the center

Have you programmed Cura to think that 0,0 is in the middle of the bed?

Dennis54, 2022-11-03:
hi there
You are not as confused as me,
I have reinstalled cura to the latest version and de installed my printers,
on the printer set-up I have set the origin as the centre of the bed.
Also upgraded the firmware of the printer to version rel 6  all axis work
well as well as bed levelling
Please find attached zip file with all info
Regards
 Dennis ( Just as confused)
Printer useless
 CR6-SE Files.zip
<https://drive.google.com/file/d/1NG-oc3jzzKf6XZrmeVV1cGR5awbIygGL/view?usp=drive_web>

Thinkersbluff, 2022-11-03:
"on the printer set-up I have set the origin as the centre of the bed."

I say again, the origin is NOT at the center of the bed. Deselect that setting in Cura.

Dennis54, 2022-11-03:
Wow, it's working, thanks for your patience.
regards
Dennis

closed 2022-11-03 by Thinkersbluff