CR6Community/Marlin #331: Adjust print speed [FR] (feature summary)

M-Janney, 2022-12-12:
### Is your feature request related to a problem? Please describe.

In previous iterations of the community software there was an option to change the print speed.  I do not see that option in the v6 community software.  Did that option go away, or is it somewhere else (or labeled something different)?

### Describe the feature you want

I would like the ability to adjust the speed of the print as per previous versions.

ritchiedc, 2022-12-12:
During a print it is next to "Flow Rate" and I believe it is called "Feed Rate" as a percentage which is consistent with Marlin terminology (M220).

M-Janney, 2022-12-12:
Ritchiedc - Thanks!  I appreciate that.  Feed rate never clicked as print speed.  In my head it registered as the rate the filament was fed to the printer, not the movement of the print head .  Off to experiment!