CR6Community/Marlin #333: CR6SE dead since firmware and screen update

Benny-H, 2023-01-14:
Screen Blank since Community Upload

So I tried to update the screen and firmware of my Kickstarter CR6SE
Definitely the old board and the original screen firmware. 4.52 and DGUS2 1.4 2019.04.10 NO MODS at all

Each time i tried to upload the screen update it would get to either the BIN file or the h2k and it would stop.
I added the Kernel update files and got the same stopping issue but when i turned it off and removed the SD card, now all i get is a blank screen.

I have since tried to do the firmware of several versions (community 6 as well as many creality versions) and it seems that the firmware is updating (6 firmware causes the led to flash) so the board is still kicking but the screen remains blank.

Help me please!

[CR6Comm-CF6-Final-cr6-se-v4.5.2-mb-2021-03-27-15-52.zip](https://github.com/CR6Community/Marlin/releases/download/v2.0.7.3-cr6-community-release-6/CR6Comm-CF6-Final-cr6-se-v4.5.2-mb-2021-03-27-15-52.zip)
CR-6 SEHW4.5.3SW2.0.1.9_20220516
CR-6SE 32bit mainboard+screen firmware new 20211112

Thinkersbluff, 2023-01-15:
Hi, Benny.

The DWIN display bootloader is not robust, and it throws no error messages, so it is very difficult for folks to debug these issues.

If you put an empty DWIN_SET folder onto a FAT32 SD card, formatted with 4096 bytes per sector, and power cycle the display, does the screen remain blank, or do you see a blue screen?

Benny-H, 2023-01-15:
I was trying that and opened a previously fitted SD card and noticed that there were only three files on the card now, the first one being the same H2k file that it stops on.
Seems that it was trying to suck down those files and stopped at that point. Still no display though, happy for any suggestions.

Ben

Thinkersbluff, 2023-01-15:
The HZK file is very large and always slow to download. You can flash that one by itself, later.

I asked whether the screen is blue or backlit and grey, when you flash, now.  

I have had my screen go grey when flashing.  Although the sd card I was using looked ok in the PC, had worked repeatedly with the display and still worked to flash the motherboard, my screen went grey when flashing with that card. I bought a new display, convinced it was failed.  Two days later, I tried a different SD card, and it flashed correctly...

That bootloader needs to read the Master Boot Record, to find the files.  If files are disappearing from a card, it is not the display doing that, it is the card.

If you have any other cards, I would try them all until one works.

Each file that you have already flashed is ok to remove from DWIN_SET, in theory. They each go into their own memory location and persist until overwritten.

You might try flashing the T5L...CFG fie by itself.  The one configures the display.  If it was not processed correctly, that could create all sorts of problems.  At the extreme, you can brick the display with the wrong T5L file.  E.g. that is the one that tells the display to look for the folder DWIN_SET and ignore everything else.

The actual graphics are in the ICL files.  The screen backgrounds are in 32.ICL. The others contain icon overlays.
The bin files 13 and 14 define the display screens and the touch maps.  
The 22 bin file contains the initial values for displayed parameters.

Benny-H, 2023-01-15:
Hey thanks again,

The screen is not blue, i am not sure if the backlight is there and i have
two SD cards now, one i bought new from Amazon and is 8gb and a Kingston
that i partitioned to 8gb.

So i can perhaps flash each file separately. Honestly it seems like it
works for a few seconds then just stops.

Ben

Thinkersbluff, 2023-01-15:
Intermittency is typically a sign of loose or broken or corroded power or ground connections.  If the screen goes black with no backlight, I would check that first!  You don't want to be interrupting power while flashing.that can brick it.

Benny-H, 2023-01-15:
It's not an intermittent issue, the screen is dead, black or only a
backlight, since flashing the kernel files.

Ben

Benny-H, 2023-01-15:
There is 4.98V at the power and ground pins to the screen.

Ben

Thinkersbluff, 2023-01-15:
This sounded like an intermittent screen? 

“Honestly it seems like it works for a few seconds then just stops.”

Benny-H, 2023-01-15:
Ahh sorry my mistake, before the screen went dead it would seem to flash
for a couple of seconds then stop. It sometimes got to the bin file and
sometimes to the H2k but never further. When I had just flashed the
community 6.1 i had the same thing. When I turned on the printer the
touchscreen would take input (in English for some reason too) then freeze.
This is what made me think i needed to start looking at flashing the screen
and got me here.

I think i messaged you on discord. I'm heading back home now and think i
know where I might have yet another SD card. If you could it might be
easier to message and try things directly.

Ben

Unamola, 2023-02-28:
Hi,
Really hope any one can help me.

So my printer had some really mess up print a few months back and the thermostat died.. i waited until i had money to spare for the repair.
This Monday i replaced the hotend with a brand new kit (https://www.amazon.es/dp/B09L16QV53?psc=1&ref=ppx_yo2ov_dt_b_product_details) and decided that well it was time to do a upgrade to the firmware has well since i had read about the community version.

So i read all and did all preparations. I know my printer is also the old board and the original screen firmware. 4.52 and DGUS2 1.4 2019.04.10
It was also a kickstart version. 
I followed a guide and it said to do the screen 1º. So i did i got a 8GB Philips Micro SD HC (https://www.amazon.es/dp/B00FEFFVRI?psc=1&ref=ppx_yo2ov_dt_b_product_details)
All went well i saw the OK mensage and power cycled the printer to redo the flash has instructed to do it 2 or 3 times.
At this moment My screen has not power on anymore, completely black. 

I have already tryed to flash with the official screen firmware from Creality website. Since screen was black i powered the printer with the SD in place and left it for 20 minutes so that it had time to run the firmware update, still the screen continues black. No Info, nothing. 

Can any one provide me some help please?

Cheers

Thinkersbluff, 2023-02-28:
Really sorry to hear the screen has gone black. That usually means some kind of power failure on or to the display. The last few folks who had this issue had to replace the display.

rockymtngeek, 2023-02-28:
@Unamola if you haven't already, take a close look at the cable for the display. I had multiple issues with mine in the beginning and found a pinch point between where it enters the case and the back of the drawer section of the printer. Make sure it can move a little freely within that space otherwise it might be pinched causing intermittent issues. I wound up replacing the entire cable due to it being damaged (over pinched) from factory assembly.

ritchiedc, 2023-02-28:
@Unamola did you include the kernel upgrade files in the dwin_set folder on the micro sd card when you flashed the screen?

Unamola, 2023-03-01:
> @Benny-H & @Unamola if you haven't already, take a close look at the cable for the display. I had multiple issues with mine in the beginning and found a pinch point between where it enters the case and the back of the drawer section of the printer. Make sure it can move a little freely within that space otherwise it might be pinched causing intermittent issues. I wound up replacing the entire cable due to it being damaged (over pinched) from factory assembly.

Will do that today, Tk u for suggesting it

Unamola, 2023-03-01:
> @Unamola did you include the kernel upgrade files in the dwin_set folder on the micro sd card when you flashed the screen?

Yes i did i followed a guide, that said to do it

Unamola, 2023-03-01:
> Really sorry to hear the screen has gone black. That usually means some kind of power failure on or to the display. The last few folks who had this issue had to replace the display.

Lets hope it doest come to that because if it does i will have to wait another month or so to do it.

Benny-H, 2023-03-02:
Hi and thanks for the comments. Checked the cables and ended up needing a new main board and screen and since been working well with the community software apart from some sometimes strange behaviour.

Sometimes at the start of a print the head simply goes to the back right of the build plate and then turns the extruder to max flow, clicking and generally sounding scary.
Other times the print will stop at random and the SD card is blank.

Both issues could be the SD Card, but it was new. Ill monitor.

Regarding the screen blank after update, i think a list of what brands and types of SD cards is required, the screen is just too sensitive to what card it is and ones that are not compatible for whatever reason has the potential to brick the screen.

Ben

Thinkersbluff, 2023-03-09:
Interesting idea, but way too many SD cards on the market to list all of the ones that work or do not work, I think.

I generally find that a surface scan of whatever card I am using will highlight any bad sector issues.  Bricking the screen is at least theoretically possible, I suppose, if a sector happens to go bad in the boot record or somewhere within a T5L cfg file. We don't think the bootloader routine is particularly robust in terms of error handling.

The advice from DWIN is to stick to 8Gb cards or less. I suppose that helps screen-out cards using extended capacity (XC) technologies and discourages using High Capacity (HC) cards (though HC cards also work for me, if I partition them to 8Gb MBR partitions).

I was once using a  both my Samsung 64Gb (XC) card repeatedly, (partitioned 15Gb MBR, formatted FAT32, 4k Sectors) to flash both my BTT SKR CR6 motherboard and my stock DWIN display. One day, the screen went blank while using that card. That card worked fine flashing my motherboard and the display stayed blank when using any of my other cards.  I ordered a replacement display, believing it had failed.  

Two days later, I flashed the "failed" display with a different SD card, and it "magically" came back to life.  Still using it.  Can't say for sure what happened or why it recovered.  Circumstantial evidence suggests the card was somehow responsible, and DWIN does warn not to use XC cards,...  

Could something inside the SD card slot become contaminated or deformed? Can locking and unlocking an SD card in the slot cause issues with the PCB?  All flimsy mechanical things can have intermittent or catastrophic failures and we know Creality like to squeeze the pennies on bulk buys like this...

Our general advice on cards is to stick to 8Gb or smaller, partition MBR (which is not the default on MACs, for instance), and format FAT32, 4096 allocation units.  The MBR, FAT32 and 4k specs are all dictated by limitations of the DWIN bootloader.  Whether XC or HC technologies can operate intermittently on that system,  even if partitioned to 8Gb or less for the first partition, I cannot say, but it's possible, even with brand name cards like Sandisk, Kingston, Samsung, etc...

NOTE: I have been assuming that a black screen is a power failure, while a blank screen is a bricked display.
It may also be possible that the bootloader routine can actually leave the screen in a black (power-off) state, which could explain why some users end up with a black screen while trying to flash firmware to these displays.  I can not say for sure, one way or the other.

closed 2023-03-13 by Thinkersbluff