CR6Community/Marlin #335: Heating Failed

Alvarojcg09, 2023-01-29:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

Whenever I try to print or heat the nozzle or bed this message shows up:
” HEATING FAILED
Turn printer on and off
Target temperature was not met on nozzle or bed soon enough
Verify ambient conditions are suitable for printing.” 

### Bug Timeline

It didn’t happened with the stock firmware 

### Version of CR6Community Firmware

4.5.2

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.2

### Display

Creality CR-6 stock touch screen

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

Thinkersbluff, 2023-01-30:
Can you switch the part cooling fan on and off with the CF firmware installed?

Alvarojcg09, 2023-01-30:
> Can you switch the part cooling fan on and off with the CF firmware installed?
> 
> The 4.5.2 firmware has worked correctly on CR6 printers with 4.5.2 boards in them for over a year, now. If the 4.5.2 firmware is installed on a printer with a 4.5.3 or 1.1.0.3 motherboard, however, the heaters will fail to operate, the part fan will fail to operate, and the optical sensor will fail to operate (meaning that the probe will plough into the bed if you try to home or level the bed.

Alvarojcg09, 2023-01-30:
Yeah, I downloaded the wrong fail. What file I have to install for a 1.1.0.3 motherboard?

Thinkersbluff, 2023-01-30:
https://github.com/CR6Community/Marlin/releases/download/v2.0.8.1-cr6-community-release-6.1/CF6.1-Final-cr6-se-v4.5.3-mb-2021-06-04-16-53.zip

closed 2023-01-30 by Thinkersbluff